autorope/donkeycar #749: Jetson Nano - Error "No module named imgaug" when starting the training

custom-build-robots, 2021-01-17:
Hi,
today I installed my jeston nano with the latest donkey car framework version. I did it by hand step by step and after that does not work I installed everythin new and used the `install-jp44.sh` script.
Driving around and collecting data worked after fixing two error.
The D`RIVE_TRAIN_TYPE = "SERVO_ESC"` was missing and the `CSIC_CAM_GSTREAMER_FLIP_PARM = 0` in the myconfig.py file.

Then I tried to start the training but no luck. I always got the errror:

```
(env) donkey@donkey:~/mycar$ python train.py --tubs=data/tub_test/,data/tub_2_21-01-16/,data/tub_1_21-01-16/,data/tub_3_21-01-16/ --model models/big.h5
________             ______                   _________
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/
                                 /____/

using donkey v4.1.0-dev ...
2021-01-17 17:25:59.245273: I tensorflow/stream_executor/platform/default/dso_loader.cc:48] Successfully opened dynamic library libcudart.so.10.2
Traceback (most recent call last):
  File "train.py", line 15, in <module>
    from donkeycar.pipeline.training import train
  File "/home/donkey/projects/donkeycar/donkeycar/pipeline/training.py", line 10, in <module>
    from donkeycar.pipeline.augmentations import ImageAugmentation
  File "/home/donkey/projects/donkeycar/donkeycar/pipeline/augmentations.py", line 3, in <module>
    import imgaug.augmenters as iaa
ModuleNotFoundError: No module named 'imgaug'
(env) donkey@donkey:~/mycar$ No module named 'imgaug'
```
Any idea whats wrong with my installation?

Thank you all very much for your help,
Ingmar

tikurahul, 2021-01-17:
We did not include `imgaug` in the list of required packages for the Nano, because we don't expect people to train on the Nano.

You need to install `imgaug` on the Nano. Unfortunately this package does not have prebuilt wheels on ARM64, which means this will take a while to compile. 

https://github.com/aleju/imgaug is the package.

custom-build-robots, 2021-01-19:
> We did not include `imgaug` in the list of required packages for the Nano, because we don't expect people to train on the Nano.
>
> You need to install `imgaug` on the Nano. Unfortunately this package [...]

Thank you @tikurahul I will try to install imgaug on the nano. At the moment I build (un-commented) the three imgaug lines in the training.py file. But it would be nice if the compiling will do the the job.

For me the problem is now understood / fixed.

closed 2021-01-19 by custom-build-robots

Ezward, 2021-12-17:
@custom-build-robots Were you able to complete the training and then use the model on your nano?  The reason I ask is that we are having trouble when installing imgaug on Jetson Nano running Jetpack 4.6.  When we install imgaug the CSIC camera stops working.  It would be great to know your configuration if you were successful.  Thanks.