autorope/donkeycar #758: Install-44jp.sh fails at line 23

TCIII, 2021-01-21:
NVIDIA Nano 4GB
OS Jetpack 4.4.1

While executing install-jp44.sh received: "Could not open requirements file: [Errno 2] No such file or directory: 'requirements.txt'" in response to line 23: "sudo -H pip3 install -r requirements.txt"

TCIII, 2021-01-21:
Installed "requirements.txt" in the root directory and commented out "update" and "upgrade: in "install-jp44.sh.
Manually ran "update" and "upgrade", then rebooted the Nano and then ran "install-jp44.sh".
I had one instance of "failed building wheel for h5py", but script did not return to the CLI.
Script continued to run until the installation of Tensorflow when I received a massive slew of errors and the script went to the CLI.

I do not believe that this installing script for the Nano is ready for use.

sctse999, 2021-01-26:
Let's put this into the doc.

JosefGst, 2021-02-09:
I ran into same errors. Is there a workaround installing donkey 4.x on the nano?

TCIII, 2021-02-09:
@JosefGst,

I successfully installed DC 4.1.0 on the Nano using the manual installation guide. The installation script is definitely flawed as noted above and should be removed from the Nano DC installation docs.

Ezward, 2023-03-04:
@TCIII what is the status here?  We don't support JP4.4 anymore, but is the underlying issue still an issue with JP 4.5.1 or JP 4.6?

TCIII, 2023-03-04:
@Ezward,

I have not had this issue when installing either JP4.5.1 or JP4.6.1/2 when using either the `main` branch or the `tf_2_9` branch which will soon be merged with the main branch.

TCIII