autorope/donkeycar #761: Missing manifest.json file?

kl2301, 2021-01-22:
I am using the Ubuntu application and I have followed the instructions on the Donkeycar documentation for Host PC setup (Linux OS) and the Train an autopilot instructions up to training a model. When I try to execute `donkey train --tub ./data/tub_9_21_01-08 --model ./models/mypilot.h5`, this error is returned.

[image]

kl2301, 2021-01-25:
Fixed, I needed to update the tub directory by using the script I found in `~/projects/donkeycar/scripts/convert_to_tub_v2.py`.

closed 2021-01-25 by kl2301

nonegom, 2021-04-05:
> Fixed, I needed to update the tub directory by using the script I found in `~/projects/donkeycar/scripts/convert_to_tub_v2.py`.

May I know how you solved it? about code

czj942650673, 2021-05-12:
how can i find the manifest.json