autorope/donkeycar #764: Web Controller is not active

mf01, 2021-01-22:
I want to try donkey car 4.1.0 with the simulator on Win10. DonkeyCar and the simulator are starting but not the web controller. Port 8887 is not open. I think this not a firewall issue because port 9091 is open and working. This is was I've done and the output.

```
python .\manage.py drive
________             ______                   _________
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/
                                 /____/

using donkey v4.1.0 ...
d:\ai-loop\donkeycar\donkeycar\donkeycar\parts\datastore.py:221: SyntaxWarning: "is" with a literal. Did you mean "=="?
  elif typ is 'image':
loading config file: D:\AI-Loop\DonkeyCar\mycar\config.py
loading personal config over-rides from myconfig.py
cfg.CAMERA_TYPE PICAM
starting DonkeyGym env
setting default: start_delay 5.0
setting default: max_cte 5.0
setting default: frame_skip 2
setting default: cam_resolution (120, 160, 3)
setting default: log_level 20
donkey subprocess started
INFO:gym_donkeycar.core.client:connecting to 127.0.0.1:9091
C:\Users\m02frank\.virtualenvs\DonkeyCar-RsYoVS50\lib\site-packages\gym\logger.py:30: UserWarning: [33mWARN: Box bound precision lowered by casting to float32[0m
  warnings.warn(colorize('%s: %s'%('WARN', msg % args), 'yellow'))
WARNING:gym_donkeycar.envs.donkey_sim:waiting for sim to start..
loading scene generated_track
INFO:gym_donkeycar.envs.donkey_sim:on need car config
INFO:gym_donkeycar.envs.donkey_sim:sending car config.
INFO:gym_donkeycar.envs.donkey_sim:done sending car config.
Adding part DonkeyGymEnv.
Adding part XboxOneJoystickController.
Adding part ThrottleFilter.
Adding part PilotCondition.
Adding part RecordTracker.
Adding part DriveMode.
Adding part AiLaunch.
Adding part AiRunCondition.
Adding part TubWriter.
You can now move your joystick to drive your car.
Joystick Controls:
+------------------+---------------------------+
|     control      |           action          |
+------------------+---------------------------+
|     a_button     |        toggle_mode        |
|     b_button     |  toggle_manual_recording  |
|     x_button     |    erase_last_N_records   |
|     y_button     |       emergency_stop      |
|  right_shoulder  |   increase_max_throttle   |
|  left_shoulder   |   decrease_max_throttle   |
|     options      |  toggle_constant_throttle |
|      circle      | show_record_acount_status |
|        R2        |      enable_ai_launch     |
| left_stick_horz  |        set_steering       |
| right_stick_vert |        set_throttle       |
|  right_trigger   |       set_magnitude       |
|   left_trigger   |       set_magnitude       |
+------------------+---------------------------+
Starting vehicle at 20 Hz
no support for fnctl module. joystick not enabled.
```

python1995, 2021-01-25:
I have the same problem. Did you manage to solve the problem ?

mf01, 2021-01-26:
No, actual I have no solution or workaround.

sctse999, 2021-01-26:
By default, when you are using a joystick, the web controller is not active.

sctse999, 2021-01-26:
Having said that, I am working on an enhancement to make the web controller turned on even when the user is using the controller.

mf01, 2021-01-26:
It is very unintuitive to uncomment USE_JOYSTICK_AS_DEFAULT and set to False. The tutorial I
've read points out that webcontroller is enabled on default.

sctse999, 2021-01-26:
I agree with you! @mf01

closed 2021-01-26 by mf01

sctse999, 2021-01-26:
https://github.com/autorope/donkeycar/pull/769