autorope/donkeycar #765: error creating donkey car

balakreshnan, 2021-01-24:
updated python and rebase with 4.1
(env) pi@autocar:~/projects/donkeycar$ python --version
Python 3.6.5
(env) pi@autocar:~/projects/donkeycar$ donkey createcar --path ~/mycar
Traceback (most recent call last):
  File "/home/pi/env/bin/donkey", line 11, in <module>
    load_entry_point('donkeycar', 'console_scripts', 'donkey')()
  File "/home/pi/env/lib/python3.5/site-packages/pkg_resources/__init__.py", line 561, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/pi/env/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2631, in load_entry_point
    return ep.load()
  File "/home/pi/env/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2291, in load
    return self.resolve()
  File "/home/pi/env/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2297, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/pi/projects/donkeycar/donkeycar/__init__.py", line 8
    print(f'using donkey v{__version__} ...')
                                           ^

tikurahul, 2021-01-24:
We don't support python 3.5 anymore. Please upgrade to 3.6 or above.

balakreshnan, 2021-01-24:
i already did python 3.6

sctse999, 2021-01-26:
@balakreshnan as you can see from the console printout above, you are using python 3.5.  If you don't mind, i am closing this issue now.

closed 2021-01-26 by sctse999