autorope/donkeycar #767: Mismatching versions?

kl2301, 2021-01-25:
I'm using the donkeycar docs at docs.donkeycar.com and I am trying to run the command `python manage.py drive --model ~/mycar/models/mypilot.h5` However, I am running into the following error.
[image]
[image]

I'm not sure where this is going wrong, but something I have noticed is that whenever I am ssh'd into my pi, the software says it is using the donkey version 3.1.5. When I am using my Host PC terminal, the software says it is using donkey version 4.1.0. Could this be related? How would I get them to be the same version?

tikurahul, 2021-01-26:
You are most definitively using the wrong version of Tensorflow.

DocGarbanzo, 2021-07-11:
Please pull from git and you should be on the latest on both platforms.

closed 2021-07-11 by DocGarbanzo