autorope/donkeycar #772: Fatal error in actuator.py when run on NVIDIA Nano

TCIII, 2021-01-27:
Nano OS: Jetpack 4.5.1
DC: 4.1.0

Setting the DRIVE_TRAIN_TYPE = "DC_TWO_WHEEL" unfortunately causes "manage.py" to call part "actuator.py" which at line 576 imports "Rpi.GPIO as GPIO" which results in the following error: "RuntimeError: This module can only be run on a Raspberry Pi!"

Obviously this is not going to work on an NVIDIA Nano which requires "Jetson.GPIO to be imported.

Unless I missed something, there appears to be no checks in "actuator.py" as to which processor, Rpi or Nano, is being used and arbitrarily uses Rpi.GPIO when a Drive train type other than the default "SERVO_ESC" is selected.

zlite, 2021-01-27:
We don't actually have a config parameter that specifies which board this is running on. Is there a way we can query the OS to find out? 

If not, we should add a cfg.PROCESSOR_TYPE to the master config

TCIII, 2021-01-27:
Good observation Chris, however the DC software had to have known the processor configuration or it wouldn't have issued the "RuntimeError: This module can only be run on a Raspberry Pi!" error.

The obvious solution to this import the default Rpi.GPIO unless the processor is a Nano and then import Jetson.GPIO

As a temporary solution is it possible for me to replace the Rpi.GPIO with Jeston.GPIO to keep my project moving along?

zlite, 2021-01-27:
Worth trying that. One way or another we'll have to have a Jetson GPIO solution

TCIII, 2021-01-27:
I found this on the Jetson.GPIO Github:

"11. PWM
See samples/simple_pwm.py for details on how to use PWM channels.

The Jetson.GPIO library supports PWM only on pins with attached hardware PWM controllers. Unlike the RPi.GPIO library, the Jetson.GPIO library does not implement Software emulated PWM. Jetson Nano supports 2 PWM channels, and Jetson AGX Xavier supports 3 PWM channels. Jetson TX1 and TX2 do not support any PWM channels.

The system pinmux must be configured to connect the hardware PWM controlller(s) to the relevant pins. If the pinmux is not configured, PWM signals will not reach the pins! The Jetson.GPIO library does not dynamically modify the pinmux configuration to achieve this. Read the L4T documentation for details on how to configure the pinmux."

The bottom line here is that unlike the Rpi GPIO Bus, the Nano GPIO Bus cannot support more than two PWM channels and there appears to be a fair amount of flag waving required to implement even those two channels.

Recommend that this issue be closed as the Nano is presently incapable of directly supporting PWM on its GPIO pins the same way as the Rpi family of SBCs and a notice added to the Nano DC docs concerning this PWM issue.

sctse999, 2021-01-28:
I didn't check if Jetson.GPIO share the same interface with Rpi.GPIO. If this is the case, what you need to do is to write a simple wrapper with a `try` block to import the correct library. Something like this:

```
try:
    import RPi.GPIO as GPIO
except:
    pass

try:
    import Jetson.GPIO as GPIO
except:
    pass
```

BrianHenryIE, 2022-03-06:
This happened me today, the solution was:

`pip uninstall RPi.GPIO`

Error was `python manage.py drive`:

```
________             ______                   _________              
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /    
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/     
                                 /____/                              

using donkey v4.3.5 ...
Traceback (most recent call last):
  File "manage.py", line 24, in <module>
    from donkeycar.parts import actuator, pins
  File "/home/brian/projects/donkeycar/donkeycar/parts/actuator.py", line 13, in <module>
    from donkeycar.parts.pins import OutputPin, PwmPin, PinState
  File "/home/brian/projects/donkeycar/donkeycar/parts/pins.py", line 397, in <module>
    import RPi.GPIO as GPIO
  File "/home/brian/env/lib/python3.6/site-packages/RPi/GPIO/__init__.py", line 23, in <module>
    from RPi._GPIO import *
RuntimeError: This module can only be run on a Raspberry Pi!
```

Ezward, 2023-02-18:
This is caused by user error when installing; using the raspberry pi instructions on a jetson nano.  Simply uninstalling Rpi.GPIO fixes the issue. 
```
pip uninstall RPi.GPIO
```

closed 2023-02-18 by Ezward