autorope/donkeycar #775: No throttle or steering control when using XBox One gamepad in local control

TCIII, 2021-01-28:
SBC: NVIDIA Nano 4GB
OS: Jetpack 4.5.1
DC: 4.10

The XBox One gamepad is paired and connected to the Nano OS and during "manage.py" drive button pushes on the gamepad show the expected response on the CLI.

I have set USE_JOYSTICK_AS_DEFAULT =  True in "myconfig.py", however when running manage.py drive, there is no steering or throttle output from the PCA9685, other than a constant 1,500 us PWM output, when I move the XBox One left joystick right or left or the right joystick forward or backwards.

A review of the "controller.py" part shows that is it missing hex keycodes for the right joystick horizontal and vertical, the left joystick vertical, and several other hex keycodes.

Even though the left joystick horizontal hex keycode is present, there was no steering output response to movement of the gamepad left joystick in the horizontal direction.

tikurahul, 2021-01-28:
The right joystick is actually not used when using the XBox One Controller. We use the shoulder buttons instead. The right trigger is accelerator, and the left trigger is braking.

TCIII, 2021-01-28:
@tikurahul,

I beg to differ with you. From the "controller.py" part:

"self.axis_trigger_map = {
            'left_stick_horz': self.set_steering,
            'right_stick_vert': self.set_throttle,
            # Forza Mode
            'right_trigger': self.magnitude(),
            'left_trigger': self.magnitude(reversed = True),

This also appears in the "manage.py drive" startup XBox One "Joystick Control" CLI output:

 left_stick_horz  |        set_steering       |
| right_stick_vert |        set_throttle       |
|  right_trigger   |       set_magnitude       |
|   left_trigger   |       set_magnitude

To not be consistent with the operation other gamepads will just lead to beginner heartburn.

tikurahul, 2021-01-28:
> To not be consistent with the operation other gamepads will just lead to beginner heartburn.

Respectfully disagree. There is almost no overlap in users who use PS5 vs XBox and game pad conventions and expectations are very different. That being said - I wrote the Xbox controller this way, because this  seemed a *lot* more intuitive to me when driving the car. You are welcome to write a configurable part that solves your use-case.

TCIII, 2021-01-28:
Will do

tikurahul, 2021-01-29:
This looks correct. You can probably update the documentation with this keymap.

Ezward, 2023-03-04:
I'm going to close this.  We have a separate issue for refactoring the game controller system to allow all functions to be remapped.  https://github.com/autorope/donkeycar/issues/1097

closed 2023-03-04 by Ezward