autorope/donkeycar #779: AttributeError: 'Config' object has no attribute 'DRIVE_TRAIN_TYPE'

rndr-man, 2021-02-01:
Donkey v4.1.0

Could not start the car until I added the following from an old config file.

DRIVE_TRAIN_TYPE = "SERVO_ESC" # SERVO_ESC|DC_STEER_THROTTLE|DC_TWO_WHEEL|SERVO_HBRIDGE_PWM|PIGPIO_PWM|MM1|MOCK

Error message if attribute is not in config:

Adding part DriveMode.
Traceback (most recent call last):
  File "manage.py", line 171, in <module>
    drive(cfg, model_path=args['--model'], model_type=args['--type'])
  File "manage.py", line 136, in drive
    if cfg.DONKEY_GYM or cfg.DRIVE_TRAIN_TYPE == "MOCK":
AttributeError: 'Config' object has no attribute 'DRIVE_TRAIN_TYPE'

sqy42, 2021-02-02:
I have exactly the same problem. If I add DRIVE_TRAIN_TYPE to myconfig.py, the car starts.

DocGarbanzo, 2021-07-11:
We have added the missing config parameter in the configs.

closed 2021-07-11 by DocGarbanzo