autorope/donkeycar #780: controller.py missing custom class

tomsepe, 2021-02-01:
I tried creating a custom joystick controller but I am getting an error exception. it seem that there is no definition for a custom class.

  File "/home/tom/projects/donkeycar/donkeycar/parts/controller.py", line 1506, in get_js_controller
    raise( Exception("Unknown controller type: " + cfg.CONTROLLER_TYPE))
Exception: Unknown controller type: custom

sqy42, 2021-02-02:
Same problem here. Generated the file `my_joystick.py` with `donkey createjs`. Changed `CONTROLLER_TYPE="custom"` in `myconfig.py` and now when I execute `python manage.py drive`, the following error is shown:

```
(env) pi@raspberry:~/mycar $ python manage.py drive
________             ______                   _________              
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /    
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/     
                                 /____/                              

using donkey v4.1.0 ...
loading config file: /home/pi/mycar/config.py
loading personal config over-rides from myconfig.py
PiCamera loaded.. .warming camera
Adding part PiCamera.
Traceback (most recent call last):
  File "manage.py", line 171, in <module>
    drive(cfg, model_path=args['--model'], model_type=args['--type'])
  File "manage.py", line 101, in drive
    ctr = get_js_controller(cfg)
  File "/home/pi/projects/donkeycar/donkeycar/parts/controller.py", line 1506, in get_js_controller
    raise( Exception("Unknown controller type: " + cfg.CONTROLLER_TYPE))
Exception: Unknown controller type: custom
```

How can I fix this?

Ezward, 2023-02-18:
This works correctly in version 4.4

closed 2023-02-18 by Ezward