autorope/donkeycar #783: How can I train my donkeycar on jetson nano

lee-junho21, 2021-02-03:
I'm using Jetpack4.4.1 on my jetson nano. And I installed donkey car 4.1 version.
After then, I could drive my donkey car with ps4 controller. But, when I try to train my donkey car, I faced a problem.
On jetson nano, I tried the train code

donkey train --tub ./mycar/data --model ./models/mypilot.h5 --type=linear

Then, Error occurs. like

AttributeError: 'NoneType' object has no attribute 'DEFAULT_AI_FRAMEWORK'

So, I added DEFAULT_AI_FRAMEWORK = None in myconfig.py
Then, import error: No module named imgaug occured.

I also tried this code on my window computer, but I had same problem.
How can I solve this problem. I really hope to train my donkeycar...

custom-build-robots, 2021-02-03:
Hi @lee-junho21,
the command you are using looks wrong to me.

Try the command for example as follows:
python train.py --tubs=data/tub_2_21-01-16/,data/tub_1_21-01-16/,data/tub_3_21-01-16/ --model models/mypilot.h5

For some more details how to train on a jetson nano (what is not the best way but it is possible) I have published a pdf file describing the setup on a jetson nano and how to train and run a neural net on the jetson nano.
[Ingmars Donkey Car – E-Book](https://custom-build-robots.com/top-story-de/donkey-car-e-book/14046)

Or visit the project in the discord channel to get some help: [discord server](https://discord.gg/PN6kFeA)

Best,

Ingmar

TCIII, 2021-08-01:
@lee-junho21,

DEFAULT_AI_FRAMEWORK = 'tensorflow'

kory2000pr, 2021-09-06:
Did you resolve the imaaug problem???

Ezward, 2023-03-04:
The library we use for image augmentations does not work correctly on Jetson Nano, so we do not install it on the Nano, that is why you are getting that error.  Unfortunately the maintainer of imgaug has said that he will not fix it.  https://github.com/autorope/donkeycar/issues/970

We have a separate issue to completely refactor the image augmentations and we can address this issue when we do that.  https://github.com/autorope/donkeycar/issues/1109