autorope/donkeycar #784: ValueError

lee-junho21, 2021-02-05:
I trained my data on my window OS , so I could get mypilot.h5. And I moved mypilor.h5 from window to jetson nano.
But, When I start autonomous driving, I faced ValueError.

I entered code 
 : python manage.py drive --model ~/mycar/models/mypilot.h5,

After the code had started, I changed my mode by using my ps4(share button) to local pilot or local angle.
Then ValueError occurs.

ValueError : Input 0 of layer dense_1 is incompatible with the layer : expected axis -1 of input shape to have value 6656 but received input with shape [None, 28224]

How can I resolve my Error? I really hope to drive my car autonomously

DocGarbanzo, 2021-02-14:
Which tensorflow version have you installed on nano? And which version of tf and DC are you running on your PC?

lee-junho21, 2021-02-15:
> Which tensorflow version have you installed on nano? And which version of tf and DC are you running on your PC?

I used tensorflow2.2.0 in nano and I also used tf2.2.0, docopt0.6.2. Is that any problem?

DocGarbanzo, 2021-02-16:
This is the right version. Can you try with `--type tensorrt_linear`?

DocGarbanzo, 2021-07-11:
Closing b/c of lack of activity.

closed 2021-07-11 by DocGarbanzo