autorope/donkeycar #788: OSError: [Errno 121] Remote I/O error whilst running "python manage.py drive"

Zayr0, 2021-02-15:
Hello, I'm stuck on this issue. I always get a remote I/O error whilst running "python manage.py drive".
I have activated i2c in the raspi-config and I also checked the wires from my pi going to the PiRacer board (i2c wires).
I can see the PCA9685 address using `i2cdetect -y 1` and I also am part of the i2c group. See below:
[image]
However, I still get this error. Any ideas/solutions? Thanks in advance!

closed 2021-02-19 by Zayr0

BossZweifel, 2021-02-26:
Hi Zayr0, I am having now the exact same problem. What was your solution?

Zayr0, 2021-02-26:
Hi BossZweifel, for me it was that the i2c address of the PCA9685 was wrong (mycar/config.py), it was 60 which didn't exist in my i2c detect. However, I still can't properly steer and give throttle at the same time, as the program does not like it when you give it address 40 twice for both (even with different channels of the PCA board). But maybe you'll figure it out, good luck!

BossZweifel, 2021-02-27:
I put both on 0x40 now, specified the bus and calibrated it properly. While calibratiting, the servo was steering and the motor was turning as specified.
The python manage.py drive starts without an error. The webinterface is there and shows the slow camera feed. 

But nothing happens when I give any inputs. I was really hoping to getting it to at least drive a bit today ...

Zayr0, 2021-02-27:
Yeah same. If you put one of them to a different address then 0x40 (but one that is detected by i2c detect) the other one does work. Maybe open it up as a separate issue on the issue page?