autorope/donkeycar #791: Since moving from v 3.1.5 to v 4.1.0 the PS4 gamepad no longer functions per the docs

TCIII, 2021-02-22:
SBC: NVIDIA Nano
OS: Jetpack 4.4.3
DC: v 4.1.0

When updating from DC v 3.1.5 to v 4.1.0, my PS4 gamepad for local control no longer functioned as it did when running v 3.1.5.

A subsequent review of the master "controller.py" part revealed that the PS4 hex keycodes are incorrect for both the right joystick H and V axes, the Dpad left/right buttons and the Dpad up/down buttons, the L2 and R2 trigger axes, the L1 and R1 buttons, the circle, the cross, and the square buttons, the share, and the options buttons. Using jstest I determined the following:

Right stick horizontal: IS: 0x03; S/B: 0x02
Right stick vertical: IS: 0x04; S/B: 0x05

Left trigger axis: IS: 0x02; S/B: 0x03
Right trigger axis: IS; 0x05; S/B 0x04

L1 button: IS: 0x138; S/B 0x134
R1 button: IS: 0x139; S/B 0x135

Circle button: IS: 0x131; S/B 0x132
Cross button: IS: 0x130; S/B 0x131
Square button: IS: 0x134: S/B 0x130
Triangle button: IS correct (0x133)

Pad button: IS correct (0x13d)
Share button: IS: 0x13a; S/B: 0x138
Options button: IS: 0x13b; S/B: 0x139
PS button: IS: correct (0x13c)

Dpad left/right buttons: IS: 0x10; S/B 0x06
Dpad up/down buttons: IS: 0x11; S/B: 0x07

BrianHenryIE, 2021-05-18:
Thanks. I was able to get driving by replacing [controller.py:457-496](https://github.com/autorope/donkeycar/blob/1c4569c4ddf7d2878c73c6af19eee8f67fca79a1/donkeycar/parts/controller.py#L457-L496) with:

```
        self.axis_names = {
            0x00 : 'left_stick_horz',
            0x01 : 'left_stick_vert',
            0x02 : 'right_stick_horz',
            0x05 : 'right_stick_vert',

            0x03 : 'left_trigger_axis',
            0x04 : 'right_trigger_axis',

            0x06 : 'dpad_leftright',
            0x07 : 'dpad_updown',

            0x19 : 'tilt_a',
            0x1a : 'tilt_b',
            0x1b : 'tilt_c',

            # 0x06 : 'motion_a',
            # 0x07 : 'motion_b',
            0x08 : 'motion_c',
        }

        self.button_names = {

            0x134 : 'square',
            0x130 : 'cross',
            0x131 : 'circle',
            0x133 : 'triangle',

            0x138 : 'L1',
            0x139 : 'R1',
            0x136 : 'L2',
            0x137 : 'R2',
            # 0x13a : 'L3',
            # 0x13b : 'R3',

            0x13d : 'pad',
            0x13a : 'share',
            0x13b : 'options',
            0x13c : 'PS',
        }
```

Incomplete. As I need to make more changes I'll edit.

Notice the repeated dictionary keys in the code in master:

https://github.com/autorope/donkeycar/blob/1c4569c4ddf7d2878c73c6af19eee8f67fca79a1/donkeycar/parts/controller.py#L489-L494

zlite, 2021-05-18:
Brian, thanks for the catch. Can you submit a PR with those changes, please?

BrianHenryIE, 2021-05-18:
I’m a little hesitant because of [this comment on Discord](https://discord.com/channels/662098530411741184/671604367706554388/827585434800488489)

> I have found that the hex keycodes for both the PS4 and XBox One are different depending on what SBC (Rpi/Nano) you are using.

I’m on Nvidia Jetson Nano.

Although with the duplicate dictionary keys, I can’t imagine it being correct for RPi either.

TCIII, 2021-05-18:
@BrianHenryIE,

The hex keycodes for the PS4 and XBox One in "controller.py" were generated with the Joystick Wizard and not jstest and are substantially different between the two hex keycode generators depending on the SBC in use.

I unfortunately tried to run Joystick Wizard on my Nano 4GB running JP 4.5.1 and immediately had a "core dump" which caused an unrecoverable error such that I had to burn a new JP 4.5.1 SD card.

Since Joystick Wizard will not run on a Nano 4GB running JP 4.5.1 without causing a "core dump" and an unrecoverable error, I believe that there should be a warning in the DC Joystick docs about this issue.

zlite, 2021-05-19:
I actually don't think it matters if a keycode has two dictionary entries.
One describes the key and the other the function. You can call either.

BrianHenryIE, 2021-05-19:
PyCharm prompted me to remove the entries with duplicate keys in the dictionary. And the Python documentation does say that dictionaries have ["the requirement that the keys are unique"](https://docs.python.org/3/tutorial/datastructures.html#dictionaries). 

I guess this is a mapping of raw input values from the joystick to friendly names, and later the friendly names are mapped to Donkey Car functions. So at this level it doesn't matter if two buttons map to the same "name", but the same physical button's binary (hex) signal shouldn't be defined twice.

Where you're saying "you can call either", I think that can be true, but in different parts of the code. Bear in mind I'm still only dipping my toe in the code.

I think this issue should remain open, particularly for visibility this week. I'm not 100% sure of things yet to open a PR (e.g. I commented out parts). I'd hoped to work on it this evening but I think I killed my Donkey Car!

zlite, 2021-05-19:
Okay, I've reopened. Please update on your testing and submit a PR with your suggested changes. We'll test it on Rpi before merging.

BrianHenryIE, 2022-05-09:
I have a branch that I think fixes this. I'm not 100% confident in it yet. Every time I reinstall Donkey I seem to have to change something. But right this moment, this config is working for me:

https://github.com/autorope/donkeycar/compare/main...BrianHenryIE:ps4-throttle-steering-fix

To test:

```
git remote add brianhenryie https://github.com/brianhenryie/donkeycar.git;
git fetch --all;
git merge brianhenryie/ps4-throttle-steering-fix;
```

I'm surprised nobody else seems to be having this issue. i.e. would merging this mess up configurations that are working for others?

Ezward, 2022-05-25:
I know @TCIII uses the PS4 on Nano.  He was having this issue.

Ezward, 2023-02-18:
Can we close this?

TCIII, 2023-02-18:
@Ezward,

Yes. The PS4 hex key codes are correct for the Rpi, but not for the Nano and required custom editing of the `controller.py` part.
The solution was to use the Joystick Wizard to create a custom joystick.
TCIII

closed 2023-02-18 by TCIII