autorope/donkeycar #795: Steering and Throttle issue i2c

BossZweifel, 2021-02-27:
Hi there,

I am using a raspberry pi 4 with the waveshare racer kit. My raspberry detects the following i2c addresses

[image]

As standard, 0x40 is used for steering, and 0x60 for throttle. As you can see, I don't have a 0x60 i2c address detected. 

If I put both on 0x40, my web interface shows everything correctly but does not let me control the car. If I change the throttle to any other available address, my steering seems to work but my throttle does nothing. 

While calibrating, the servo and motor were fully functional. This leads me to believe there is a problem with the i2c configuration.

Does anybody have a solution for this? 
Thanks in advance!

TCIII, 2021-03-03:
@BossZweifel

First off "PCA9685_I2C_BUSNUM =" should be set to "None" for the Rpi.
Second, I doubt that the DC software is setup to work with two PCA9685s (such as ADDR1) at the same address.
You might want to pose your issue [here](https://discord.com/channels/662098530411741184/671604048218292233).

zlite, 2021-03-03:
Yes, I think you can use two devices with the same I2C address as long as they're on different buses, which are determined by which pins you connect them to on the Pi or Nano. (I2C1 and I2C2)

BossZweifel, 2021-03-06:
Updating to donkey v4.x helped in my case. I can now steer and throttle in the web interface. Seems like the old version is bugged. 
Thanks for your help

closed 2021-03-06 by BossZweifel