autorope/donkeycar #807: Do not execute "donkeycar createjs" on an NVIDIA Nano

TCIII, 2021-03-13:
Processor: NVIDIA Nano 4GB
OS: Jetpack 4.5.1
DC: 4.1.0/4.2.0 (dev)

When I attempted to execute "donkey createjs", during some gamepad troubleshooting, I encountered this error:

(env) thomas@thomas-desktop:~/mycar$ donkey createjs
________             ______                   _________
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/
                                 /____/

using donkey v4.1.0 ...
Illegal instruction (core dumped)
(env) thomas@thomas-desktop:~/mycar$

Jstest is installed and works fine with my XBox One gamepad.

After this core dump incident, I was unable to delete and recreate "mycar", with either the basic or the complete template, with out getting the dreaded "core dump" message at the start of the create process.
It appears that the attempted execution of "donkeycar createjs" caused a permanent "core dump" to occur anytime the create function is invoked.

To see if the core dump issue was permanent, I completed the following:

1) I burned Jetpack 4.5.1 onto a 128GB SD card and then proceeded to install the necessary software, per the DC docs, on the 4GB Nano to run DC.
2) DC v 4.1.0 installed without issue.
3) I then proceeded to create mycar, using the complete template, which was created without the core dump error I was experiencing after attempting to run donkeycar createjs.
4) I then deleted mycar complete and created mycar using the basic template without an issue.

Moral of the story here is not to attempt to run donkeycar createjs on the Nano as it will apparently cause a permanent "core dump" when attempting to use the create function, but does not seem to affect the installation of DC software.

Ezward, 2022-01-10:
@TCIII can you include the coredump with this issue, it will help us debug it.

TCIII, 2022-01-10:
@Ezward,

> @TCIII can you include the coredump with this issue, it will help us debug it.

I have long since moved on from JP 4.5.1 so I can no longer duplicate the issue. Also, the "coredump" caused a situation where I had to totally reburn the JP 4.5.1 SD card because the Nano refused to boot up after the "coredump" incident.

TCIII

Ezward, 2022-02-28:
I have now used the joystick wizard successfully on the Jetson Nano 2gb and Jetson Nano 4gb running Jetpack 4.6.  I"m thinking the original issue was a Jp 4.5.1 issue only.  I guess we can close this issue once we are no longer support JP 4.5.1

Ezward, 2022-05-25:
@TCIII I believe you have successfully used the Joystick wizard on the Jetson Nano.  Is that correct?  If so I'd like to close this issue.