autorope/donkeycar #815: Cannot train Donkeycar

ender18g, 2021-03-16:
If I try the train command, I get the follow mmap error:

Any ideas what I could be doing wrong?

Total params: 817,028
Trainable params: 817,028
Non-trainable params: 0
__________________________________________________________________________________________________
None
Using catalog /content/mycar/data/data/catalog_27.catalog
Traceback (most recent call last):
  File "/content/mycar/train.py", line 28, in <module>
    main()
  File "/content/mycar/train.py", line 24, in main
    train(cfg, tubs, model, model_type)
  File "/content/donkeycar/donkeycar/pipeline/training.py", line 101, in train
    dataset = TubDataset(cfg, all_tub_paths)
  File "/content/donkeycar/donkeycar/pipeline/types.py", line 78, in __init__
    for tub_path in self.tub_paths]
  File "/content/donkeycar/donkeycar/pipeline/types.py", line 78, in <listcomp>
    for tub_path in self.tub_paths]
  File "/content/donkeycar/donkeycar/parts/tub_v2.py", line 26, in __init__
    read_only=read_only)
  File "/content/donkeycar/donkeycar/parts/datastore_v2.py", line 254, in __init__
    self.current_catalog = Catalog(last_known_catalog, read_only=self.read_only, start_index=self.current_index)
  File "/content/donkeycar/donkeycar/parts/datastore_v2.py", line 144, in __init__
    self.seekable = Seekable(self.path.as_posix(), line_lengths=self.manifest.line_lengths(), read_only=read_only)
  File "/content/donkeycar/donkeycar/parts/datastore_v2.py", line 25, in __init__
    self.file = mmap.mmap(self.file.fileno(), length=0, access=mmap.ACCESS_READ)
ValueError: cannot mmap an empty file

DocGarbanzo, 2021-03-16:
@ender18g, can you add a comment how you solved it - maybe your tub was empty? Might be useful for other people who see the same error.

tikurahul, 2021-03-16:
`ValueError: cannot mmap an empty file` is the error.

My guess is that the tub did not exist.

ender18g, 2021-03-17:
The mmap error was a result of trying to train multiple tubs of data at once. Within my data directory, I had 4 tubs, which I wanted to all be included in my model training.

However, when I used the following command:
`donkey train --tub ./data/* --model ./models/mypilot.h5`
I got the ValueError: cannot mmap an empty file

The only way I could avoid getting the error was to train each tub, one by one.
`donkey train --tub ./data/tub_1_2020_03_16 --model ./models/mypilot.h5`

So, I got rid of the error, but according to the donkeycar docs, ./data/* should be a usable command to train using all tubs.

Also, none of the tubs were empty.

qasimalik134, 2021-04-13:
@ender18g  Could you please share your dataset for training.

ender18g, 2021-04-13:
Thanks for the responses.  After some more investigation, it appears that all directories are indeed tub directories.  However, the 1st tub (tub_1_.... ) was an empty directory. The other tubs had content. 

The easy fix to is to simply find and delete the empty.  I'm not exactly sure what causes an empty tub, but it could be visiting the donkey car web address and then refreshing the page without driving/producing any data.

closed 2021-04-13 by ender18g