autorope/donkeycar #820: Error encountered when verifying tensorflow on Rpi 4B

TCIII, 2021-03-22:
SBC: Rpi 4B 4GB
OS: Buster Desktop
DC: v 4.1. from Master

I have run into a problem installing tensorflow during the installation of DC dependencies/modules on an Rpi 4B.
When I installed DC on a Rpi 3B+ a couple of weeks ago I had no problem installing the DC support software (dependencies/modules).
However I am installing the DC support software on a Rpi 4B and have received this error when verifying the tensorflow installation:

"from tensorflow.core.framework import function_pb2
  File "/home/pi/env/lib/python3.7/site-packages/tensorflow/core/framework/function_pb2.py", line 7, in <module>
    from google.protobuf import descriptor as _descriptor
ModuleNotFoundError: No module named 'google.protobuf'"

I have verified that the "protobuf" module has been installed:

"Requirement already satisfied: setuptools in /home/pi/env/lib/python3.7/site-packages (from protobuf) (54.1.2)
Requirement already satisfied: six>=1.9 in /usr/lib/python3/dist-packages (from protobuf) (1.12.0)"

However, it looks like the import of tensorflow is looking for a "google.protobuf" module that has not been installed yet.

I have also found that the latest version of numpy (pip install numpy --upgrade) 1.20.1 does not work with tensorflow 2.3.1 for some reason.
Numpy must be greater than 1.15.1 but less than 1.19.1. for tensorflow 2.3.1.

DocGarbanzo, 2021-07-11:
@TCIII - can you check you still have these problems using master?

TCIII, 2021-07-11:
I no longer have an issue with tensorflow when using DC 4.2.1 on my 4B.

closed 2021-07-11 by TCIII