autorope/donkeycar #850: Logitech G27 steering wheel use different axe (0 in stead of 2) for steering -> should be configerable (Gamepad API)

joostaafjes, 2021-04-24:
I am using a Logitech G27 steering wheel and found out that it using axes 0 for steering in stead of 2 as used currently [main.js](https://github.com/autorope/donkeycar/blob/dev/donkeycar/parts/web_controller/templates/static/main.js#L293)

It would be nice to make this axes configurable (preferable in config.py/myconfig.py) but than this value should be propagated to the main.js in some way.

Created this little tool to find the HTML5 gamepad key mapping: https://github.com/joostaafjes/html5-gamepad-tester

sctse999, 2021-05-17:
Interesting. Can you submit a PR? I will look into this.

Ezward, 2021-06-12:
@joostaafjes we do have a joystick mapper.  See `Joystick Wizard` in [docs](https://docs.donkeycar.com/utility/donkey/).  It is a general facility for mapping any game controller that mounts on `/dev/input/js0`.  You create a mapping, then use `CONTROLLER_TYPE="custom"` in you config to use it.

TCIII, 2021-06-12:
@Ezward,

Executing Joystick Wizard on a Nano 4GB running JP 4.5.1 will cause an unrecoverable core dump due to an illegal instruction.

See issue [#807](https://github.com/autorope/donkeycar/issues/807) 

Also, Joystick Wizard does not create the same hex keycodes as jstest does on a Nano.

Ezward, 2023-03-04:
I believe @TCIII you have successfully used the Joystick Wizard on the Jetson Nano.  I have also.  So that is a work-around.  Also, we intend to refactor the game controller system in a way that would allow a user to implement support for a new or unique controller simply by mapping the codes that come out of it.  https://github.com/autorope/donkeycar/issues/1097

So I am going to close this issue.

closed 2023-03-04 by Ezward