autorope/donkeycar #854: exceptions when trying to start training for the first time with v4.1.0

JohannesHoh, 2021-04-28:
Im getting two types of exceptions.

Please note that im a first time user of donkey car but not a pyhton / opencv newbie... So i may just use the donkeycar wrong...

1) Following the documentation i try to run training by using the command `donkey train --tub ./data/data* --model ./models/mypilot.h5`
```

$  donkey train --tub ./data/data* --model ./models/mypilot.h5
________             ______                   _________
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/
                                 /____/

using donkey v4.1.0 ...
loading config file: ./config.py
loading personal config over-rides from myconfig.py
Traceback (most recent call last):
  File "C:\Apps\Python39\Scripts\donkey-script.py", line 33, in <module>
    sys.exit(load_entry_point('donkeycar', 'console_scripts', 'donkey')())
  File "c:\dev\donkey\projects\donkeycar\donkeycar\management\base.py", line 477, in execute_from_command_line
    c.run(args[2:])
  File "c:\dev\donkey\projects\donkeycar\donkeycar\management\base.py", line 442, in run
    framework = args.framework if args.framework else cfg.DEFAULT_AI_FRAMEWORK
AttributeError: 'Config' object has no attribute 'DEFAULT_AI_FRAMEWORK'

```

2) second try after figuring out that the docu is maybe out of date and i need to pass a the --framework param

```
$  donkey train --tub ./* --model ./models/mypilot.h5 --framework pytorch
________             ______                   _________
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/
                                 /____/

using donkey v4.1.0 ...
loading config file: ./config.py
loading personal config over-rides from myconfig.py
Traceback (most recent call last):
  File "C:\Apps\Python39\Scripts\donkey-script.py", line 33, in <module>
    sys.exit(load_entry_point('donkeycar', 'console_scripts', 'donkey')())
  File "c:\dev\donkey\projects\donkeycar\donkeycar\management\base.py", line 477, in execute_from_command_line
    c.run(args[2:])
  File "c:\dev\donkey\projects\donkeycar\donkeycar\management\base.py", line 448, in run
    from donkeycar.parts.pytorch.torch_train import train
  File "c:\dev\donkey\projects\donkeycar\donkeycar\parts\pytorch\torch_train.py", line 5, in <module>
    from donkeycar.parts.pytorch.torch_data import TorchTubDataModule
  File "c:\dev\donkey\projects\donkeycar\donkeycar\parts\pytorch\torch_data.py", line 9, in <module>
    from donkeycar.pipeline.sequence import TubSequence
  File "c:\dev\donkey\projects\donkeycar\donkeycar\pipeline\sequence.py", line 45, in <module>
    class TfmIterator(Generic[R, XOut, YOut],  SizedIterator[Tuple[XOut, YOut]]):
  File "c:\apps\python39\lib\abc.py", line 85, in __new__
    cls = super().__new__(mcls, name, bases, namespace, **kwargs)
TypeError: Cannot create a consistent method resolution
order (MRO) for bases Generic, SizedIterator
`
```

And fyi - just in case the problem may be here: my data folder looks like this after i copied the data from the raspberry pi to my windows machine: 
```

$ ls data
catalog_0.catalog  catalog_0.catalog_manifest  catalog_1.catalog  catalog_1.catalog_manifest  catalog_2.catalog  catalog_2.catalog_manifest  data/  images/  manifest.json

```

JohannesHoh, 2021-04-30:
Update: same for donkey v4.2.0

JohannesHoh, 2021-05-02:
Fixed it by switching to conda installation with the windows.yml env files, which is way more precise than the native pip installation instructions. I would propose to double check the windows setup in the docu, or at least leave a clear recommendation to the conda setup.

Root cause of the reported issue: Problem seems to be in the setup and has absolutely nothing to do with the training data, that i collected - the exception is quite confusing in that manner, as it could be interpreted in a way that there may be issues with the images sizes, which of course is wrong by reading through the stack trace - but a user with little experience can be easily mislead here..

closed 2021-05-02 by JohannesHoh

DocGarbanzo, 2021-05-02:
@JohannesHoh your argument should be`—tub ./data`