autorope/donkeycar #859: python3 manage.py drive has no response

YoYoHC, 2021-05-07:
cant drive the car 
System prompt  

using donkey v4.2.0 ...
Usage:
    manage.py drive [--model=<model>] [--type=(linear|categorical|tflite_linear)]

What shall I do?

DocGarbanzo, 2021-05-09:
Please post the full shell output of your command and the return.

YoYoHC, 2021-05-20:
> Please post the full shell output of your command and the return.

(env) pi@raspberrypi:~ $ cd mycar
(env) pi@raspberrypi:~/mycar $ python3 manage.py driver
________             ______                   _________              
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /    
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/     
                                 /____/                              

using donkey v4.2.0 ...
Usage:
    manage.py drive [--model=<model>] [--type=(linear|categorical|tflite_linear)]
(env) pi@raspberrypi:~/mycar $

DocGarbanzo, 2021-05-21:
Please use `python3 manage.py drive`

closed 2021-05-21 by DocGarbanzo