autorope/donkeycar #863: Problem with training model on Donkey Car simulator

OumarDiop, 2021-05-13:
When training the model for the virtual donkey car, I get  the error `problems with loss graph: [Errno 2] No such file or directory: 'C:\\Users\\coolDawg1234\\Desktop\\donke\\projects\\donkeycar\\OumarCar\\models\\models\\example_model.png'` Has anyone had this problem or knows a solution? I followed this [guide,](https://docs.donkeycar.com/guide/simulator/) on Windows 10.

NevGithub0823, 2021-05-22:
from my understanding, in 4.2, they changed it so you have to give the path that's relative to `mycar/models`. so you should change `--model models/example_model.h5` to `--model example_model.h5`. it worked for me.

OumarDiop, 2021-05-22:
Oh yeah I see that it says `models\\models`, I should've noticed that. I'll try it right now, thanks.

DocGarbanzo, 2021-06-03:
This has accidentally sneaked into 4.2, we have PR #873 to re-instate the old behaviour. The docs are reflecting what we have in master now. I'll close this, please re-open if you see still an issue.

closed 2021-06-03 by DocGarbanzo