autorope/donkeycar #870: Donkey Sim "Not Responding"

NevGithub0823, 2021-05-22:
Windows 10.
Donkey Sim works fine, but when I click the "Quit" button, it freezes and shows "Not Responding". Please help.

Maximellerbach, 2021-05-24:
Hi ! What version of the sim are you using ?
Just checked v21.04.15 and it works fine.

In order to close the app, you can also close just like a regular app using the top right close button of the window: 
[image]

NevGithub0823, 2021-05-24:
I reinstalled the sim, and now the output says there's a
```
ValueError: Scene name mountain_track not in scene list ['generated_road', 'warehouse', 'sparkfun_avc', 'generated_track', 'roboracingleague_1', 'waveshare', 'mini_monaco', 'warren', 'thunderhill']
```
and
```
RuntimeError: cannot join current thread
```
The Quit button works before the Errors (1-2 seconds after the window opens, the error comes and the Quit button causes Not Responding).

NevGithub0823, 2021-05-24:
I am also using that version of donkey sim.

Maximellerbach, 2021-05-24:
Please chose a scene from that list, mountain_track isn't in this version of the sim, that's why it's throwing you an error. For the "Not responding" part, I will take a look at it

NevGithub0823, 2021-05-24:
> Please chose a scene from that list
I didn't choose mountain_track, it just says I did, idk why

Maximellerbach, 2021-05-27:
It's in your config file, you should find a variable named "GYM_ENV" or something like that, you need to change it to a valid track name

NevGithub0823, 2021-05-28:
please tell me which one? 
[image]

Maximellerbach, 2021-05-28:
There is a list of env name in there, please choose any from it, for example: "donkey-generated-track-v0"

[image]<https://user-images.githubusercontent.com/68718784/119929881-3813ef80-bf33-11eb-8c97-99807042b4ed.png>

NevGithub0823, 2021-05-28:
thanks, it fixed that error but the window is still not responding after I press quit.
also, "X" button also makes it do that... the only thing that works for me now is to force quit the app or to just ^C in powershell.

Maximellerbach, 2021-05-28:
Do you disconnect the client before closing the sim ?

Maximellerbach, 2021-05-28:
Try to close the script and then close the sim, not the other way around

Maximellerbach, 2021-06-01:
Please tell me if that worked, if it works, let's close this issue

NevGithub0823, 2021-06-01:
it works

closed 2021-06-01 by NevGithub0823