autorope/donkeycar #876: The delete button on the controller cannot be deleted continuously.

naisy, 2021-06-01:
I recorded 1000 records and then tried to delete it by hitting the delete button repeatedly.
Only the first 20 could be deleted. (I'm using the value of DRIVE_LOOP_HZ here.)
The same range is subject to deletion every time.
range(1008, 1028)

cat manifest.json
```
["cam/image_array", "user/angle", "user/throttle", "user/mode"]
["image_array", "float", "float", "str"]
{}
{"created_at": 1622505857.6851628, "sessions": {"all_full_ids": ["21-06-01_0"], "last_id": 0, "last_full_id": "21-06-01_0"}}
{"paths": ["catalog_0.catalog", "catalog_1.catalog"], "current_index": 1028, "max_len": 1000, "deleted_indexes": [1024, 1025, 1026, 1027, 1008, 1009, 1010, 1011, 1012, 1013, 1014, 1015, 1016, 1017, 1018, 1019, 1020, 1021, 1022, 1023]}
```
This seems to be caused by the management of fast_index and last_index not considering the delete flag.
https://github.com/autorope/donkeycar/blob/dev/donkeycar/parts/tub_v2.py#L77
```
    def delete_last_n_records(self, n):
        last_index = self.manifest.current_index
        first_index = max(last_index - n, 0)
        self.manifest.delete_records(range(first_index, last_index))
````

Thank you.

DocGarbanzo, 2021-06-01:
This is addressed in #877.

DocGarbanzo, 2021-06-03:
This has now been merged to dev - @naisy, can you check it works for you now?

closed 2021-06-03 by DocGarbanzo

naisy, 2021-06-04:
@DocGarbanzo 
Yes, I checked it.
It works as expected. Thank you very much.