autorope/donkeycar #882: Donkey Car cannot be updated if the controller.py part has been modified (customized).

TCIII, 2021-06-12:
SBC: Nano 4GB
OS: JP 4.5.1
DC: 4.1.0

I have had to modify (customize) the PS4 Joystick class hex keycodes in the controller.py part to obtain correct operation of the PS4 gamepad with my Nano 4GB.
When I attempted to update to DC 4.2.1, I received a message that the update was aborted because I had customized the controller.py part.
The workaround was to completely delete the mycar and donkeycar directories and reinstall donkeycar 4.2.1 from scratch. 

Would it be possible to allow DC to be updated even with a modified part through a YES or NO request as to continuing with the update instead of aborting it?

DocGarbanzo, 2021-06-12:
Git does not allow to merge (or pull) into your sandbox if you have modified files. If `controller.py` has wrong code, can you create a PR with the fix?

TCIII, 2021-06-12:
@DocGarbanzo,

Thankyou for your response, much appreciated.

Unfortunately Joystick Wizard does not generate the same hex keycodes, that appear in controller.py, for the PS4 as jstest does when using the Nano 4GB and the jstest hex keycodes are the only ones that work for the Nano.
The default hex keycodes for the PS4 in controller.py probably work for the Rpi, but definitely not for the Nano 4GB has been my experience. Therefore I have to create a modified (custom) PS4 Joystick class in controller.py so there is no fix short of creating a controller.py part specifically for the Nano.
Also, executing Joystick Wizard (donkeycar createjs) on a Nano 4GB running JP 4.5.1 will cause an unrecoverable core dump due to an illegal command (see issue [#807](https://github.com/autorope/donkeycar/issues/807)) so that a custom Joystick class cannot be created.

DocGarbanzo, 2021-06-16:
@TCIII - could you add a flag in the joystick controller classes constructor that can switch between nano and RPi and then add a different hex code mapping for corresponding platform? And submit that as a PR?

TCIII, 2021-06-17:
@DocGarbanzo,

Unfortunately I am not an experienced Python programmer.

My suggestion to handle this hex keycode issue with the Nano is a follows:
1) Add warning: Do not attempt to use the Joystick Wizard to create a "custom" joystick as running "donkey createjs" on the Nano will cause an unrecoverable core dump (see issue [#807](https://github.com/autorope/donkeycar/issues/807)).
2) Instead, install and run jstest to determine the correct gamepad hex keycodes for the Nano.
3) Use nano to edit and modify the desired gamepad hex keycodes in the "controller.py" part.
4) Verify that the gamepad in use will control the vehicle steering, throttle, etc. as expected.

DocGarbanzo, 2021-06-17:
@TCIII 
I suggest the following:
1. We'll add this to the docs
2. Please upload your nano version of the file & I will code the change & submit a PR
3. Obsolete after 2
4. Obsolete after 2

If you could test the change when available on dev - that would be great.

TCIII, 2021-06-18:
@DocGarbanzo,

Per your request I am attaching my "controller.py" part that has the PS4Joystick class modified to work with the NVIDIA Nano and Xavier NX. The hex keycodes are based on those generated by jstest. The PS4Joystick class works for both the PS4 and the PS5.  
[controller.zip](https://github.com/autorope/donkeycar/files/6678113/controller.zip)

Ezward, 2023-02-18:
@TCIII I believe we resolved this issue by using the Joystick Wizard to create a custom mapping for you.  Is that correct?  In any case I would like to close this issue as won't do; modifying something other than the files in the mycar folder must be done in a fork or branch.  So that fact that you could not update is as designed; it is a source control issue.

Ezward, 2023-02-18:
This is an issue of source control.

closed 2023-02-18 by Ezward