autorope/donkeycar #883: Updating DC to 4.2.1 will break the Robo HAT MM1 configuration

TCIII, 2021-06-12:
SBC: Nano 4GB
OS: JP 4.5.1
DC: 4.1.0
HAT: Robo HAT MM1 M4

Issue:
When attempting to update DC 4.1.0 to 4.2.1, DC would no longer communicate with the Robo HAT MM1 over the serial bus.

Solution:
I had to delete the mycar and donkeycar directories and do a fresh install of DC 4.2.1 from scratch.

DocGarbanzo, 2021-06-12:
You can just git pull into your donkeycar directory. No need to delete it. But you have to delete the contents of your car directory, either by removing it or running the `createcar` with a flag.

TCIII, 2021-06-12:
@DocGarbanzo,

Thanks for the response, much appreciated.

TCIII, 2021-06-13:
@DocGarbanzo,

I had the same issue with all three of my Rpis too (two 3B+ and one 4B 4GB).

Ezward, 2023-02-18:
@TCIII we are at donkey 4.4 now.  Can we close this issue?

TCIII, 2023-02-18:
@Ezward,
Yes, as my Robo HAT MM1 M4 works fine with DC 4.4.4.

TCIII

closed 2023-02-18 by TCIII