autorope/donkeycar #893: Further improve tub data storage

DocGarbanzo, 2021-07-11:
# Tub data improvements

# Current state
We are storing data in the tub in read/write efficient catalog files but images are still raw jpegs inside the image directory. These can become big and hard to browse when we have too many records.

# Proposal / Advantages
* Store the images with the data, so only one file gets written 
* I've written a small routine to write the jpeg binary stream as base64 char stream instead of jpeg files, this already improved the write speed by about 1/3.
* There are probably better options to combine binary and text (json) data into one file, this should be explored.

sctse999, 2021-07-12:
All data (meta and binary) on one single file is probably superior in terms of performance. 

On the other hand, it would be more intuitive if the images could be directly inspected on the filesystem. For some users performance isn’t their main concern.

Can we make this option configurable so user can choose? Also I assume this change is transparent to other existing parts? 

Thanks for the proposal.

DocGarbanzo, 2021-07-12:
> All data (meta and binary) on one single file is probably superior in terms of performance.
>
> On the other hand, it would be more intuitive if the images could be directly inspected on the [...]

Access to the raw image data as jpeg file is a valid argument. But the value is limited if the folder gets too big. The code change I mention above supports both versions, so yes, this could be made a user defined choice. Performance will become important if you record while inferencing, because currently this steals like 8ms or so in your vehicle loop.

tikurahul, 2021-07-12:
> I've written a small routine to write the jpeg binary stream as base64 char stream instead of jpeg files, this already improved the write speed by about 1/3.

Let's no do this. This is horrible for performance. We are much better off using something like protos so we have a strong typing in our payload and we can use regular tools to diagnose when things go wrong.

We could make it configurable, but its hard when we mix tubs of different types. So that is something we would need to handle gracefully.

Ezward, 2023-03-04:
I can see how writing a single file, even if it is bigger that the total of two files, could be faster.  Given that we are writing to a MicroSD card. a single write with sequential data would probably be faster even if it was a little larger.  @DocGarbanzo have you actually measured the performance of this correct and is i an improvement?

The other consideration here is; how would this affect other tooling like tubplot or tubclean.

DocGarbanzo, 2023-03-04:
I haven’t measured the performance of writing to a single file vs multiple files, a single file of a big tub could become very large. I’m not sure if this is a good thing, because we also have to copy the file from the car to the PC, etc. Tubplot is a reading operation and tubclean - or the equivalent deletion of records in the UI - are very fast, as they only write an index of deleted files to the manifest. There would be no issue with these.