autorope/donkeycar #907: Dev branch install fails during a clean install on latest Buster Desktop

TCIII, 2021-07-23:
SBC: Rpi 4B
OS: Latest Buster Desktop

I just did a clean install of the latest Buster Desktop on a Rpi 4B 4GB and then proceeded to install DC. 
When I attempted to install DC v 4.3 (dev) I immediately received a python setup error and installation was terminated. 

(env) pi@raspberrypi:~/projects/donkeycar $ git checkout dev
Already on 'dev'
Your branch is up to date with 'origin/dev'.
(env) pi@raspberrypi:~/projects/donkeycar $ pip install -e .[pi]
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/pi/projects/donkeycar
    ERROR: Command errored out with exit status 1:
     command: /home/pi/env/bin/python3 -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/pi/projects/donkeycar/setup.py'"'"'; __file__='"'"'/home/pi/projects/donkeycar/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-e5tehb5u
         cwd: /home/pi/projects/donkeycar/
    Complete output (7 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/home/pi/projects/donkeycar/setup.py", line 4, in <module>
        from donkeycar import __version__
      File "/home/pi/projects/donkeycar/donkeycar/__init__.py", line 2, in <module>
        from pyfiglet import Figlet
    ModuleNotFoundError: No module named 'pyfiglet'
    ----------------------------------------
WARNING: Discarding file:///home/pi/projects/donkeycar. Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

So I removed the donkeycar directory and installed DC from the master branch which proceeded without error. 
I then removed the donkeycar directory and installed DC from the dev branch which proceeded without error.

Ezward, 2021-08-14:
I have duplicated this bug on the dev branch;,`pip install -e .[pi]` fails with `No module named 'pyfiglet'`. Master branch does not have this problem; it installs correctly

```
(env) pi@raspberrypi:~/projects/donkeycar $ source env/bin/activate
(env) pi@raspberrypi:~/projects/donkeycar $ pip install -e .[pi]
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/pi/projects/donkeycar
    ERROR: Command errored out with exit status 1:
     command: /home/pi/projects/donkeycar/env/bin/python -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/home/pi/projects/donkeycar/setup.py'"'"'; __file__='"'"'/home/pi/projects/donkeycar/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-m10ghjx8
         cwd: /home/pi/projects/donkeycar/
    Complete output (7 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/home/pi/projects/donkeycar/setup.py", line 4, in <module>
        from donkeycar import __version__
      File "/home/pi/projects/donkeycar/donkeycar/__init__.py", line 2, in <module>
        from pyfiglet import Figlet
    ModuleNotFoundError: No module named 'pyfiglet'
    ----------------------------------------
WARNING: Discarding file:///home/pi/projects/donkeycar. Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
WARNING: You are using pip version 21.2.2; however, version 21.2.4 is available.
You should consider upgrading via the '/home/pi/projects/donkeycar/env/bin/python -m pip install --upgrade pip' command.
```

TCIII, 2021-08-14:
@Ezward,

It is interesting that you received the same error as I did as zlite told me that he had no problem doing a fresh install of the dev branch.
Can you do a pull to correct the dev branch or is it worth doing as I have heard that the dev branch will soon be merged into a new 4.x master?

Ezward, 2021-09-10:
I figured this out.  setup.py imports `__version__` from donkeycar; that causes `__init__.py` to be loaded, which tries to import pyfiglet.  We can't import from donkeycar in setup.py because it is not yet setup.  We will have to do versions another way.  Short term fix is to set the version directly on [line 28](https://github.com/autorope/donkeycar/blob/3dddaca49acc9857182338bcfd15d64fcdb907a7/setup.py#L28) of setup.py and delete the import on [line 4](https://github.com/autorope/donkeycar/blob/3dddaca49acc9857182338bcfd15d64fcdb907a7/setup.py#L4)

Ezward, 2021-09-11:
Fixed with the PR; https://github.com/autorope/donkeycar/pull/935  

-  installation was failing because setup was trying to import from donkeycar before it was setup.
-  setup.py imports version from donkeycar; that causes init.py to be loaded,  which tries to import pyfiglet.
-  We can't import from donkeycar in setup.py because it is not yet setup
-  this fix uses pkg_resources.get_distribution() to  get the version set in setup.py and make it  available to the donkeycar code.

From now on we set the version in setup.py; it will be imported into the donkeycode. This avoids the circular setup issue.

closed 2021-09-11 by Ezward