autorope/donkeycar #908: Autopilot ImportError: /usr/lib/aarch64-linux-gnu/libgomp.so.1: cannot allocate memory in static TLS block

AntonioCervantes, 2021-07-23:
Hello, I'm using DonkeyCar 4.2.1 with a Jetson Nano 4GB and got an Import error when trying to run the autopilot using this command: `python manage.py drive --model=models/pilot_21-07-23_0.h5` 
Has anyone else ran into this issue?

I got this error: `ImportError: /usr/lib/aarch64-linux-gnu/libgomp.so.1: cannot allocate memory in static TLS block`

[image]

[image]

fengye, 2021-07-24:
This can workaround it. I think there was a link on the donkeycar docs page but can't find it now but here you go: https://github.com/keras-team/keras-tuner/issues/317#issuecomment-640181692

DocGarbanzo, 2021-08-06:
@AntonioCervantes - did @fengye's reply solve the issue for you?

AntonioCervantes, 2021-08-07:
Yes, thank you @fengye and @DocGarbanzo I was able to resolve the issue by typing the following command into the Jetson Nano's terminal:
`export LD_PRELOAD=/usr/lib/aarch64-linux-gnu/libgomp.so.1`

closed 2021-08-07 by AntonioCervantes