autorope/donkeycar #910: Integrate 2D Lidar data into Donkeycar

Ezward, 2021-08-01:
This is a discussion issue.  The topic is how to integrate 2D lidar data, as from an RPLidar or YDLidar, into the DonkeyCar framework.  It is likely that this will result in a number of following-on issues to actually implement the design that we decide upon.   I'll start the discussion by adding a strawman as the first comment in this ticket.

Ezward, 2021-08-01:
# Integrating 2D Lidar data into Donkeycar

## The 2D Lidar data

A spinning lidar captures data in 360 degree arc around it.  It provides each data point as a distance from the lidar and the angle at which that reading was taken.  It has some maximum working range.  That range can be thought of as the radius of a circle where the lidar is at the center of that circle.  Beyond that range we may actually get some readings, but we treat those a noise.

The lidar's 'zero' angle is relative to itself and so is all the data that it provides to us (it's super narcissistic).  But we care about where the points are relative to the vehicle that the lidar is mounted on.  If we treat directly forward as 'zero' angle for the vehicle, then we need to know the offset of the zero angle on the lidar relative to the forward direction of the vehicle.  We can then adjust the angle component of each data point provided by the lidar to turn it into an angle relative to the vehicle.  It may well be that the way the lidar is mounted that the zero angle of the lidar exactly coincides with forward on the vehicle, but we should include this offset (which in that case would be zero) anyway so we have a more general model that can be applied to other lidars that may not have their zero in the same place.

## Plotting the 2D Lidar data
So once we turn the lidar data into a distance and angle relative to the vehicle, then we can use trigonometry to calculate (x,y) coordinates relative to the vehicle; the vehicle is defined as being at (0,0).  

## A complication

Another thing to think about is the skew inherent in the data taken from a moving vehicle.  Remember that the each data point provided by the lidar is given as a distance from the lidar and the angle at which the lidar was pointing when it took the reading.  But if the lidar is on a moving vehicle, then the position from which each point is taken is changing.  The lidar has some rate at which it can make a full 360 degree sweep.  This is typically 5 to 10 hz.  If the vehicle is stopped, then all data points are from a circle with the same center.  However, as the vehicle moves, each point is taken from a different location.  If the vehicle is moving 1 meter per second, which is pretty slow for a race car, and the lidar is spinning at 5hz, then the lidar has moved 1/5 of a meter between taking the first point in a sweep and the last point.  There is additional skew that happens if the vehicle is turning.  

## A solution

If we know the vehicle's position and orientation as it moves, then we can simply adjust each point provide by the lidar using the vehicle's position and orientation at the time that point was taken.  

If we have good odometry, then we can apply a kinematic model that estimates the vehicles position and orientation as it moves. Such estimates are good over short distances and we only need them to be good over a meter or so.  If we have wheel encoders or an encoder on the drive motor, then we can use those.  If we have an IMU, then we can also use that for the same purpose.  Having both would be ideal.

But what if we don't have good odometry?  That is typical of a Donkeycar.  We know the throttle value and steering value, but we don't actually know the speed of the car.  However, with a little calibration we can roughly estimate the speed of the vehicle from the throttle value.  The calibration would involve taking some measurements over a known distance at different constant throttle values and measuring the time to traverse the distance.  Then we can do interpolation to estimate the speed.  That would provide a very rough estimate.  It is affected by the particular surface the vehicle is on and by the battery charge level.  If the vehicle is changing speeds, then this model also does not take into account the lag in acceleration and deceleration.  However, it is better than not doing any adjustment of the data.  

We also need to calibrate the steering angle as well; measure the radius of a full right and full left turn; then we can interpolate to estimate turn angle given a steering value.

With those two things in place, we can then use the speed and steering angle as input to a kinematic model to estimate the vehicle's relative position and orientation over time.

## A Donkeycar 2D Lidar pipeline
Acquiring the data from the 2D lidar and transforming it into data useful to the CNN involves integrating a number of new parts into the vehicle pipeline.  2D Lidar pipeline involves 4 parts that work to gather data from the 2D Lidar, adjust for the changing position of the vehicle and render it to a bitmap so it can be saved to the tub.

1. Encoder Part:
If we have a motor encoder, then this part reads the motor encoder and writes the value and the time in nanoseconds at which the data was read.  Donkeycar does have an [encoder.py](https://github.com/autorope/donkeycar/blob/dev/donkeycar/parts/encoder.py) part that assumes that we have an arduino or similar microcontroller to read the encoder.  That part defines how encoder data is written to the pipeline.  We should try to use that if we can.

2. Kinematics Part:
This part takes as input the encoder reading (if we have an encoder), the throttle value and the steering value.  It uses the encoder reading if available or throttle otherwise, and the steering value along with other relevant configuration and calibration values and maintains an estimate of the vehicle's position and orientation.  It outputs [... cut ...]
- In an indoor setting with walls (think a school), the 2D lidar alone could probably be used to calculate a decent network.  That would be a cool demo; create a network that drives around the perimeter of a room or the outside of a bulding by following a wall.  You could use it to solve a maze using wall following.

This approach does require another calibration step; calibrating the camera.  This is a well know process and there exist plenty of code samples in Python to generate the necessary calibration data.  Further, there is plenty of Python code out there to apply this calibration data to turn a camera image into a bird's eye view; this is a common process in many robotics projects.

Heavy02011, 2021-08-01:
Great preparation, Ed!

After digging a bit into lidars during the iros2020 race and currently trying to find the bug when trying to save sim lidar data into tubs (SIM_RECORD_LIDAR = True) I would like to add the following feature requests and offer to help:

1. Make the use of the lidar identical for the physical & sim world by establishing an identical pipeline / data format for lidar (I have a RP A2M8 at hand to test real & sim training)
2. ability to set position of cameraA, cameraB and lidar independent from each other. I would like the camera position a bit more up to avoid cropping images, like the camera position of the Old Guy Bots. Currently offset_x,y,z allow only one position, because they are collapsed into one position in TcpCarHandlers.cs in sdsim, as far as I experienced when trying to set them differently.
3. I hacked together a very crude part for detecting lanes and undistorting images to a top down view some time ago, much refactoring  needed before using. https://gist.github.com/Heavy02011/07320c09c1870ef1e3a00878801fdfc9

Maybe we can have a look to the Autoware.auto pipeline, https://www.apex.ai/autoware-course, see Lecture 7 | Object Perception: LIDAR, as well.

TCIII, 2021-08-01:
@Ezward,
Thanks for initiating the discussion ticket, much appreciated. 

I presently have three Slamtec RPLidars available to support this effort: A1M8-R5, A1M8-R6 (latest firmware) and a A2M8 (latest firmware). I have one running on a Rpi 4B, one running on a Nano 4GB,  and the A2M8 running on a Xavier NX. 
The lidar.py part (DC dev branch) works well sending distance data to the TubWriter based on a lowest to highest angel value data stream and is limited to a 90 - 180 degree forward field of view. Please note that the A1 versions start the 0 degree point at the motor end of the Lidar chassis and scans clockwise towards the turret end. The A2 starts the scan at the cable end of the mount and scans clockwise.
We might want to look at Dirk Prange's "Example" of sensor integration in the "Create your own Model" [guide](https://docs.donkeycar.com/dev_guide/model/) as a start?

TCIII, 2021-08-01:
@Ezward,

I really like option B as it would allow me to operate my robot chassis without the need for lanes, as in a traditional DC environment, and allow for autonomous movement around a outside or inside perimeter.

TCIII, 2021-08-01:
@Ezward,
Leave this open to useful links:

Paper on Lidar/camera [fusion](https://arxiv.org/ftp/arxiv/papers/1710/1710.06230.pdf)

Heavy02011, 2021-08-02:
fyi: Maxime did an update on gym-donkeycar for solving this: “ ability to set position of cameraA, cameraB and lidar independent from each other.” I will test this today.

Ezward, 2021-08-02:
@Heavy02011 thanks for those links.  I see in the gist that you are using the trapezoidal method for warping the image perspective; I think we should allow for that method as it is pretty easy to configure.  We can also allow the use of camera calibration matrix if the user has calibrated their camera as that also eliminates other distortions (like those from a wide angle lens).

TCIII, 2021-08-02:
> @Heavy02011 thanks for those links. I see in the gist that you are using the trapezoidal method for warping the image perspective; I think we should allow for that method as it is pretty easy to [...]

@Ezward,

How will this affect depth cameras like the Intel D435i depth camera as I would like to use the Slamtec A2M8 with that depth camera?
Also, how is the data from your realsense435i.py part integrated into the DC training pipeline?

Ezward, 2021-08-04:
In terms of the data format that is generated by the 2D Lidar part, I think it makes sense to save the all the data on each frame, even though we may only be getting some of it on each frame.  What I mean here is that because the lidar will stream in data in segments are it collects it, we will constantly be reading some subset of the 360 degrees.  However, we want to maintain a full array data with each frame we save to the tub.

So that means we want  to stream data in and update any prior data as the new data comes in.  The donkey pipeline is running 20hz or more and the lidar is only running somewhere between 4hz to 10hz. We don't want to wait for a whole 360 degrees of data before updating the data.  I think most 2D lidars will stream data and tell you the angle range that it represents, which will be some subset of the 360 degrees arc.  

The ROS message for 2D lidar is here: http://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/LaserScan.html .  The nice thing about this format is that it can hold some subset of the 360 degree arc or it can be used to hold all 360 degrees.

I can also see that we may want to tell the part that we only care about data from 0 to 180 degrees, for instance, so we are not writing more data that we will actually use.  Again, we can still update this range as it streams in.  We would just ignore data from 180 to 360 degrees in that case.

Ezward, 2021-08-04:
> How will this affect depth cameras like the Intel D435i depth camera as I would like to use the Slamtec A2M8 with that depth camera?
> Also, how is the data from your realsense435i.py part integrated into the DC training pipeline?

I would hope camera and lidar would be totally orthogonal; they should not interfere with each other.  Of course if you have both of these, you may have a bunch of redundant data and so are writing more data then you need, but it should work.  Short of some underlying driver incompatibilities anyway.  

I'm going to try the 2d lidar without the D435; with just a normal RPi 160 degree camera.  But I don't see any reason why the D435 and RPLidar would interfere; unless we define RPLidar part as a camera; since we only configure one camera at a time, so the 2D Lidar part should not be defined as a camera.  

Rather I think we should have a new kind of part for 2d planar lidar, then we can implement config and code for the various vendors.  So for cameras we have `CAMERA_TYPE`; maybe we can add a config for `2D_LIDAR_TYPE`.  Here I would create values for families of 2D lidars that share a basic driver api and underlying data format.  For instance, if all of your RPLidar models speak to the same driver and produce the same data format but the data may differ by :
* how dense the readings are,  
* how fast the  lidar may spin,
* the max and min distance range is
If those are the only difference, then we can treat those as one lidar type, like  `2D_LIDAR_TYPE = RPLIDAR_2D`.  For YD lidar models, we would have `2D_LIDAR_TYPE = YDLIDAR`.  I have an Hitachi-LG LDS-LFDS2, so config would be `2D_LIDAR_TYPE = LDS`.

We would  have additional configuration common to all 2d lidars.  `2D_LIDAR_MIN_DISTANCE`, `2D_LIDAR_MAX_DISTANCE` in meters; which would be important for filtering out data.  We probably also need configuration for angle range we want to capture; `2D_LIDAR_MIN_ANGLE`, `2D_LIDAR_MAX_ANGLE` in radians; again go we can filter out data we don't care about.  

I'm not sure how to handle configuring the number of samples we can expect in that desired range; that is determined by how fast the Lidar is spinning and the number of readings per rotation.  I believe readings per rotation are generally fixed, but the motor speed is commonly adjustable via PWM.  So perhaps we have config for `2D_LIDAR_PPR` (pulses per rotation), which will depend upon your specific lidar.  If we put that in configuration, the we can allow the driver to determine the best spinning rate and then calculate the number of readings it needs to save based on pulses per rotation and the spin rate it chooses.

TCIII, 2021-08-04:
@Ezward,

Yes, I was going to recommend that you look at zlite's lidar.py part to see how the data capture range is limited and how many Lidars are presently supported.
The Slamtec A1M8 configuration sets the 0 degree point at the motor end of the Lidar and proceeds in a clockwise sweep towards the turret end and then back to the motor starting point. The A2M8 configuration on the other hand has a brushless motor in the base and sets the ) degree point at where the cable exits the base and sweep clockwise around the base.
This [link](https://guitar.ucsd.edu/maeece148/index.php/2019SpringTeam1) may help visualize how the A1M8 scans.

Ezward, 2021-08-07:
> @Ezward,
> 
> Yes, I was going to recommend that you look at zlite's lidar.py part to see how the data capture range is limited and how many Lidars are presently supported. Unfortunately this data capture range limiting only works for the A1M8 that references the zero degree start point of the turret scan at the back of the chassis where the motor is located.
> The Slamtec A1M8 configuration sets the 0 degree point at the motor end of the Lidar chassis The A2M8 sets the 0 degree point at the cable end of the Lidar base and proceeds in a clockwise sweep around the base back to the cable end starting point.

I think we can detect the zero point on any 2D lidar; we simply look at that last reading we saved; if the current reading's angle is less than the previous reading's angle, then we have crossed the zero point.  We can double-buffer scan; when we cross zero we start a new buffer and move the last buffer (which should be a full 360) to a 'last-buffer' status.  Further, I think we can merge these two buffers in a way that gets us a single 360 degree scan, which may include points from the latest scan and points from the previous scan, but no redundant angles.  So I don't think we actually need the zero point provided by the RPLidar, but we do need to know where the lidar considers zero degrees relative to 'forward' so we can adjust the output of all lidars so they have zero as forward.

TCIII, 2021-08-07:
@Ezward,

Even though there is a YDLidar class in the lidar.py part, zlite told me that it was not presently functional.
The Slametc A1 and YD X4 Lidars are probably the only two reasonably priced Lidars around with both the A2M8 and YD G4 touching over $300 plus tax on Amazon.

TCIII, 2021-08-08:
@Ezward,

Lidar and Camera data fusion GitHub links:

[Lidar and Camera Fusion for 3D Object Detection based on Deep Learning for Autonomous Driving](https://github.com/ZiningWang/Sparse_Pooling/blob/master/avod/BDD_Lidar-camera%20Fusion.md)

[CLOCs: Camera-LiDAR Object Candidates Fusion for 3D Object Detection](https://github.com/pangsu0613/CLOCs)

[SFND 3D Object Tracking](https://github.com/yasenh/lidar-camera-fusion-ttc)

[Sensor Fusion NanoDegree- Camera Course](https://github.com/udacity/SFND_Camera) Looks to be the best

[lidar-camera-fusion](https://github.com/topics/lidar-camera-fusion) Lidar/Camera calibration

Ezward, 2021-08-10:
Here are a couple of libaries that can manipulate point clouds and do a few other things (like scene segmentation)
- https://pointclouds.org/
- http://www.open3d.org/

Ezward, 2022-10-03:
Discussion of Lidar odometry and deskewing lidar data (skewing that happens when scan data is acquired on a moving platform).  https://youtu.be/9FhKgAEQTOg  Here is the github for the KISS ICP library https://github.com/PRBonn/kiss-icp  This includes Python bindings.