autorope/donkeycar #912: Can use 2D lidar if using a RealSense D435 camera

Ezward, 2021-08-04:
If you try to configure `CAMERA_TYPE="D435"` and `LIDAR_TYPE="RP"` simultaneously, the D435 part will be correctly started when manage.py is run, but the RPLidar will not.  RPLidar works fine is camera is `MOCK`.

The problem is a bug in `manage.py` (created from `donkeycar/templates/complete.py`).  The code that configures that camera for some reason includes the lidar config.  So on line 90 of https://github.com/autorope/donkeycar/blob/dev/donkeycar/templates/complete.py we start configuration of the camera with the 'stereo' camera, then on line 113 we 'elif' camera is "D435", then on line 136 we 'else' the other cameras.  That 'else' block also contains the lidar configuration.  So if we configure the D435 then the else is never executed and so the lidar part is not configured.

Solution: move the code block starting on line 166, https://github.com/autorope/donkeycar/blob/5e2c5608fb6b1158300266613d10807369085e56/donkeycar/templates/complete.py#L166, to outside and after the else, so it gets executed no matter what kind of camera is being configured.

TCIII, 2021-08-04:
@Ezward,

This solution only works if you have not created mycar.

I believe that we can modify manage.py in the mycar directory to correct this issue?

TCIII, 2021-08-04:
@Ezward,

Okay, as a quick fix I moved the # add lidar code to after "V.add(cam, inputs=inputs, outputs=outputs, threaded=threaded)" and on the same indent as the "else" in manage.py and here is the result:

(env) thomas@thomas-desktop:~/mycar$ python manage.py drive
________             ______                   _________
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/
                                 /____/

using donkey v4.3.0 ...
loading config file: /home/thomas/mycar/config.py
loading personal config over-rides from myconfig.py
INFO:__main__:PID: 10323
cfg.CAMERA_TYPE D435
INFO:__main__:cfg.CAMERA_TYPE D435
Depth Scale is:  0.0010000000474974513
INFO:donkeycar.vehicle:Adding part RealSense435i.
adding RP lidar part
INFO:donkeycar.vehicle:Adding part RPLidar.
Starting Donkey Server...... you can now go to thomas-desktop.local:8887 to drive your car.
INFO:donkeycar.vehicle:Adding part LocalWebController.
INFO:donkeycar.vehicle:Adding part PS4JoystickController.
INFO:donkeycar.vehicle:Adding part ThrottleFilter.
INFO:donkeycar.vehicle:Adding part PilotCondition.
INFO:donkeycar.vehicle:Adding part RecordTracker.
INFO:donkeycar.vehicle:Adding part DriveMode.
INFO:donkeycar.vehicle:Adding part AiLaunch.
INFO:donkeycar.vehicle:Adding part AiRunCondition.
PWM Steering created
Init ESC
PWM Throttle created
INFO:donkeycar.vehicle:Adding part PWMSteering.
INFO:donkeycar.vehicle:Adding part PWMThrottle.
INFO:donkeycar.vehicle:Adding part TubWriter.
You can now move your joystick to drive your car.
Joystick Controls:
+------------------+--------------------------+
|     control      |          action          |
+------------------+--------------------------+
|      share       |       toggle_mode        |
|      circle      | show_record_count_status |
|     triangle     |   erase_last_N_records   |
|      cross       |      emergency_stop      |
|        L1        |  increase_max_throttle   |
|        R1        |  decrease_max_throttle   |
|     options      | toggle_constant_throttle |
|        R2        |     enable_ai_launch     |
| left_stick_horz  |       set_steering       |
| right_stick_vert |       set_throttle       |
+------------------+--------------------------+
Opening /dev/input/js0...
Device name: Wireless Controller
INFO:donkeycar.vehicle:Starting vehicle at 30 Hz
recorded 10 records
recorded 20 records
recorded 30 records
recorded 40 records
recorded 50 records
recorded 60 records
recorded 70 records
^CINFO:donkeycar.vehicle:Shutting down vehicle and its parts...
INFO:donkeycar.vehicle:Part Profile Summary: (times in ms)
INFO:donkeycar.vehicle:
+-----------------------+-------+------+-------+-------+-------+-------+-------+
|          part         |  max  | min  |  avg  |  50%  |  90%  |  99%  | 99.9% |
+-----------------------+-------+------+-------+-------+-------+-------+-------+
|     RealSense435i     |  1.40 | 0.14 |  0.24 |  0.21 |  0.33 |  0.55 |  1.24 |
|        RPLidar        | 14.69 | 0.02 |  0.70 |  0.59 |  0.84 |  3.47 | 11.32 |
|   LocalWebController  |  0.17 | 0.03 |  0.04 |  0.03 |  0.04 |  0.09 |  0.16 |
| PS4JoystickController |  0.91 | 0.03 |  0.10 |  0.09 |  0.15 |  0.44 |  0.90 |
|     ThrottleFilter    |  0.14 | 0.01 |  0.02 |  0.02 |  0.02 |  0.07 |  0.13 |
|     PilotCondition    |  0.23 | 0.01 |  0.01 |  0.01 |  0.01 |  0.04 |  0.15 |
|     RecordTracker     |  8.85 | 0.01 |  0.05 |  0.01 |  0.02 |  0.58 |  5.48 |
|       DriveMode       |  0.16 | 0.01 |  0.02 |  0.02 |  0.02 |  0.06 |  0.11 |
|        AiLaunch       |  0.29 | 0.01 |  0.02 |  0.02 |  0.02 |  0.05 |  0.18 |
|     AiRunCondition    |  0.09 | 0.01 |  0.01 |  0.01 |  0.01 |  0.05 |  0.08 |
|      PWMSteering      |  0.21 | 0.03 |  0.04 |  0.03 |  0.04 |  0.10 |  0.20 |
|      PWMThrottle      |  0.09 | 0.01 |  0.02 |  0.02 |  0.02 |  0.06 |  0.08 |
|       TubWriter       | 51.98 | 8.87 | 21.88 | 20.09 | 32.10 | 40.39 | 50.82 |
+-----------------------+-------+------+-------+-------+-------+-------+-------+
(env) thomas@thomas-desktop:~/mycar$ ^C

As you can see the TubWriter is taking an average of 21.88ms to write both the D435i andthe RPLidar which is probably way too long. My loop is running at 30Hz and not 20Hz so maybe going back to 20Hz might help somewhat.
Also I am using a Xavier NX SBC with a U3 speed 128GB micro SD card.
As a comparison, with just the D435i camera enabled, the average write time is 7.24ms and with just the RPLidar enabled, the average write time is 4.11ms.

DocGarbanzo, 2021-08-04:
@TCIII - can you check what the tubwriter performance is if you do not write the lidar data into the tub, while keeping everything else the same? I would be surprised if the additional data had any impact on the writing speed.

TCIII, 2021-08-04:
@DocGarbanzo,

The TubWriter is taking an average of 21.88ms to write both the D435i andthe RPLidar which may be too long. My loop is running at 30Hz and not 20Hz so maybe going back to 20Hz might help somewhat.
Also I am using a Xavier NX SBC with a U3 speed 128GB micro SD card.

As a comparison, with just the D435i camera enabled, the average write time is 7.24ms and with just the RPLidar enabled, the average write time is 4.11ms.

Ezward, 2021-08-04:
@TCIII I'm hoping you can do a pull request that I can code review and merge via github.  This will get you into the flow of contributing.  The process is;

- Fork the github repo.  You can do this from the github page https://github.com/autorope/donkeycar .  In the upper right there is a 'Fork' section; click on that and it will create a version of the repo in _your_ github account.  
- Clone your forked version using from _your_ github account to your computer (using the `Code` button on the repo page).
- checkout the dev branch, `git checkout dev`
- make sure you have the latest `git pull dev`
- make your changes to `donkeycar/template/complete.py`; that is the file that becomes manage.py when you do the `donkey createcar` command. 
- test it (create a new car and make sure it works as you want)
- commit and push your changes to your branch.
- go to the main donkeycar github repo and create a "New Pull Request" using the button (or you might see that there is a message showing your changes in your fork and offering to create a pull request, you can use that shortcut if it is offered).
- do the pull request from your branch in your fork to the dev branch in the autorope/donekeycar repo.
- then let me know and I will code review the pull request.

Here is a video that describes the process; https://www.youtube.com/watch?v=a_FLqX3vGR4

If you run into any issues, let me know.

TCIII, 2021-08-04:
@Ezward,

I edited the complete.py template from my branch as required with GitHub Desktop, but I cannot figure out how to get it back to my branch so I can test it.

TCIII, 2021-08-04:
@Ezward,

Okay, I deleted my present donkeycar and mycar directories and download donkeycar from my fork.
Donkeycar installed without issue.
I then created mycar and edited myconfig.py as required.
I then ran manage.py drive successfully without receiving any errors.
Both the D435i and the RPLidar parts ran without issue.

Time for a pull request.

DocGarbanzo, 2021-08-05:
Fixed in #913 - which is now merged.

closed 2021-08-05 by DocGarbanzo