autorope/donkeycar #920: Donkey Train error "noneType" object has no attribute "models_path"

MrWhale-1, 2021-08-19:
Hey guys anyone help with this issue? 

donkey train script not working

[image]

cfox570, 2021-08-19:
Cannot find the file config.py What directory did you run from?   You need to start the train script from ~/mycar.  Config.py is in the mycar directory.

MrWhale-1, 2021-08-22:
Thanks for your reply. Can you provide some more help? I really want to get this working

I changed the directory to cd ~/mycar on my PC. Do I need to navigate the the directory instead on my Pi? 

next I did: donkey train --tub ./data/ --model ./models/mypilot.h5

then I get an error in the attached image
[image]

In my data folder I have a bunch of catalog and images
[image]

Maybe the issue is I should ssh into my pi and run train from there?

cfox570, 2021-08-23:
The training doc is very clear. You must train on the PC.  [DonkeyCar Training](https://docs.donkeycar.com/guide/train_autopilot/ )The error above says that specific image file is not readable.   It is not easy to delete a single file since there are references in the catalog files.   I suggest you make a new small set of images and run through the training process to validate it.   You may want to verify that all your images are non-zero length.   In the future, attach higher resolution screen shots, these are barely readable!   I also recommend you ask for help on Discord where you will reach a wide audience. Find the link here [Community](https://www.donkeycar.com/community.html)