autorope/donkeycar #924: In both the DC master and dev branches there is no "test_encoder.py" in the tests folder

TCIII, 2021-08-22:
In the DC "encoder.py" part introduction there is a reference to a "test_encoder.py code in the Donkeycar tests folder.

A review of the DC tests folder in both the master and dev branches found no "test_encoder.py" test.

 However a "test_encoder.py" test can be found here: https://github.com/zlite/donkeycar/tree/master/donkeycar/tests

Suggest that either a "test_encoder.py" test be added to both the DC master and dev tests directory or that the "test_encoder.py" code reference be removed from the DC "encoder.py" part introduction until such time that the "test_encoder.py" test is added to both DC branches.

Ezward, 2021-09-22:
We intend to deprecate the old encoder code in favor of tachometer.py and odometer.py in this branch; https://github.com/autorope/donkeycar/tree/921-next-generation-odometer-parts  That code does have tests.  When that get's integrated into master we can close this issue.  I would not otherwise bother addressing this issue.

Ezward, 2023-02-18:
PR https://github.com/autorope/donkeycar/pull/1089 introduces a new encoder based pipeline for encoder->tachometer-odometer-kinematics pose estimation.  tachometer.py has it's own test program build into the file's __main__.  The older encoder.py implementation is marked as deprecated, so I am closing this issue.

closed 2023-02-18 by Ezward