autorope/donkeycar #925: In the complete.py template mm_per_tick is hard coded with the value 0.306096 in the "HAVE_ODOM" cfg.

TCIII, 2021-08-23:
The robot wheel that I have an encoder attached to is 6.5 inches in diameter for a circumference of  20.42 inches. The encoder wheel has 20 slots so there are 1.02 inches/tick or 25.93 mm/tick. Therefore one rotation of the wheel should equal 518.69 mm and 10 rotations of the wheel should equal 5,185.94  mm or a little over 5 meters.

If I start manage.py drive and rotate the wheel 10 rotations while collecting records, I consistently get  between 0.06 - 0.07 meters of travel after shutting down "manage.py drive" when I should have a little over 5 meters.

I can set "MM_PER_TICK =" in myconfig.py and I am using 25.9347. In the RotaryEncoder class the default is "mm_per_tick=0.306096".

No matter what value I set for "MM_PER_TICK =" in either myconfig.py or the RotaryEncoder class, the distance travelled is always 0.06 m for 10 turns of the wheel when it should be a little over 5 meters.

If I do the following: (0.306096 mm/tick X 20 ticks/revolution) X 10 revolutions, I get 61.21 mm or 0.0612 meters of travel.
If I do the following: (0.306096 mm/tick X 20 ticks/revolution) X 5 revolutions, I get 30.61 mm or 0.0306 meters of travel.

The 0.0612 meters is approximately the distance I get when I run manage.py drive to collect records and turn the wheel 10 times. The 0.0306 meters is approximately the distance I get when I run manage.py drive to collect records and turn the wheel 5 times.

Therefore I believe that the code is using "mm_per_tick=0.306096" no matter what value is input in MM_PER_TICK =" in myconfig.py.

TCIII, 2021-08-23:
Ed Murphy reviewed the HAVE_ODOM cfg in the complete.py template and found that "mm_per_tick=0.306096" is hard coded in "enc = RotaryEncoder(mm_per_tick=0.306096, pin = cfg.ODOM_PIN, debug = cfg.ODOM_DEBUG)".

Recommended solution:

Change "mm_per_tick=0.306096" to "mm_per_tick=cfg.MM_PER_TICK"

Ezward, 2021-08-24:
I started working on this.  the pin input and the poll delay are also not being set correctly.  The other encoder (Arduino) also does not get its mm_per_tick set correctly.   One takes throttle as in input (so it can set direction of the tick change correctly for a single-channel encoder), but does not use it.  There is an encoder driver in another file; https://github.com/autorope/donkeycar/blob/dev/donkeycar/parts/pigpio_enc.py, that only outputs ticks.  there is a separate part to take in the ticks and output odometry.  So I think we want to rationalize these parts.  I started working on the two parts in encoder.py and saw they have different ways of calculating the distance and speed; basically they repeat the same basic logic, but diverge.  This would not happen if they only output ticks and the odometry part took that in and calculated distance and speed.  So, I think the right thing to do is to simplify these parts to only output ticks and elevate the odometry part to it's own file.  It will take in the encoder ticks and derive distance and speed from them; so this logic will be in one place and separate from any hardware considerations.  So this in more than a one liner, but still relatively small.  It's a good refactor.

TCIII, 2021-08-24:
@Ezward,

Sounds good to me.

Please let me know if I can be of any assistance vetting the new encoder code and any additional supporting parts.

Ezward, 2021-09-04:
Fixed by https://github.com/autorope/donkeycar/issues/925

closed 2021-09-04 by Ezward