autorope/donkeycar #926: Performance issue in /donkeycar/parts (by P3)

DLPerf, 2021-08-25:
Hello! I've found a performance issue in donkeycar/parts/salient.py: `with tf.Session() as session`[(here)](https://github.com/autorope/donkeycar/blob/b1af0aacf488c938231912f89c4bfa5c2b11c649/donkeycar/parts/salient.py#L87) is repeatedly called in the loop `for layer in [5, 4, 3, 2, 1]`[(here)](https://github.com/autorope/donkeycar/blob/b1af0aacf488c938231912f89c4bfa5c2b11c649/donkeycar/parts/salient.py#L74).

`tf.Session` being defined repeatedly could lead to incremental overhead. If you define `tf.Session` out of the loop and pass `tf.Session` as a parameter to the loop, your program would be much more efficient.

Looking forward to your reply. Btw, I am very glad to create a PR to fix it if you are too busy.

DLPerf, 2021-11-04:
Hello, I'm looking forward to your reply~

sctse999, 2021-11-04:
Yes pls do a PR. If you can submit a test case which can show the performance gap it would be fantastic.