autorope/donkeycar #953: Update stop sign detector to run without EdgeTPU

Ezward, 2021-11-19:
Code is using this model:
https://github.com/google-coral/edgetpu/raw/master/test_data/ssd_mobilenet_v2_coco_quant_postprocess_edgetpu.tflite

So you could use that same ssd mobilenet v2 coco in a more generic tflite variable, then modify the code run it on the RPi or Nano.
See https://medium.datadriveninvestor.com/mobile-object-detector-with-tensorflow-lite-9e2c278922d0
and
https://www.tensorflow.org/lite/examples/object_detection/overview

Ezward, 2022-10-14:
It looks like Tensorflow has changed their approach to object detection; https://www.tensorflow.org/lite/examples/object_detection/overview  So we might want to use the Tensorflow Object Detection API https://github.com/tensorflow/models/tree/master/research/object_detection

Ezward, 2023-02-26:
Note on performance; remember that we are also running the normal autopilot model, so performance described above will be low because we are trying to run multiple models.  No one has done this as yet, so we don't know what the performance would be.  
- RaspberryPi: The original stop sign detector used a USB connected Coral TPU; that is still an option and relatively cheap at about $50.  
- Jetson Nano: Nano users generally run the autopilot using a TensorRT model on the GPU.  If you also want to do object/person detection, then you are better off running a tflite autopilot model on the cpu and a TensorRT object detection model on the GPU; in that way you can run the heavy model on the GPU.

Ezward, 2023-02-26:
Ok, now Google has deprecated https://github.com/tensorflow/models/tree/master/research/object_detection, and now recommends https://github.com/tensorflow/models/tree/master/official/vision Google sucks.  This thing looks harder to use, perhaps just go back to the original mobilenet suggestion in the initial issue description.

Ezward, 2023-02-27:
A user on [discord](https://discord.com/channels/662098530411741184/671604367706554388/1079799420914258021) also has used this [tensorflow-lite example](https://github.com/tensorflow/examples/tree/master/lite/examples/object_detection/raspberry_pi) for people detection on small images (160x120) on the RaspberryPi

cfox570, 2023-05-19:
See Pull Request StopSign and Cone Detection https://github.com/autorope/donkeycar/pull/1131
I used TensorFlow Lite Object Detection with a custom cone model to implement traffic cone avoidance. It was tested with Raspberry Pi 4. I am adding a pull request for the docs Create object_detection.md https://github.com/autorope/donkeycar/pull/51