autorope/donkeycar #956: manage.py drive failing

beeplove, 2021-11-28:
`python manage.py drive` is failing with the following error:

```
using donkey v4.2.1 ...
loading config file: /home/beeplove/mycar/config.py
loading personal config over-rides from myconfig.py
INFO:root:PID: 7160
INFO:root:cfg.CAMERA_TYPE CSIC
cfg.CAMERA_TYPE CSIC
INFO:donkeycar.vehicle:Adding part CSICamera.
Adding part CSICamera.
Starting Donkey Server...... you can now go to jetson-nano.local:8887 to drive your car.
INFO:donkeycar.vehicle:Adding part LocalWebController.
Adding part LocalWebController.
INFO:donkeycar.vehicle:Adding part ThrottleFilter.
Adding part ThrottleFilter.
INFO:donkeycar.vehicle:Adding part PilotCondition.
Adding part PilotCondition.
INFO:donkeycar.vehicle:Adding part RecordTracker.
Adding part RecordTracker.
INFO:donkeycar.vehicle:Adding part DriveMode.
Adding part DriveMode.
INFO:donkeycar.vehicle:Adding part AiLaunch.
Adding part AiLaunch.
INFO:donkeycar.vehicle:Adding part AiRunCondition.
Adding part AiRunCondition.
PWM Steering created
Init ESC
PWM Throttle created
INFO:donkeycar.vehicle:Adding part PWMSteering.
Adding part PWMSteering.
INFO:donkeycar.vehicle:Adding part PWMThrottle.
Adding part PWMThrottle.
Traceback (most recent call last):
  File "manage.py", line 719, in <module>
    meta=args['--meta'])
  File "manage.py", line 681, in drive
    tub_writer = TubWriter(tub_path, inputs=inputs, types=types, metadata=meta)
  File "/home/beeplove/projects/donkeycar/donkeycar/parts/tub_v2.py", line 112, in __init__
    self.tub = Tub(base_path, inputs, types, metadata, max_catalog_len)
  File "/home/beeplove/projects/donkeycar/donkeycar/parts/tub_v2.py", line 28, in __init__
    read_only=read_only)
  File "/home/beeplove/projects/donkeycar/donkeycar/parts/datastore_v2.py", line 249, in __init__
    self._read_contents()
  File "/home/beeplove/projects/donkeycar/donkeycar/parts/datastore_v2.py", line 325, in _read_contents
    self.inputs = json.loads(self.seekeable.readline())
  File "/usr/lib/python3.6/json/__init__.py", line 354, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3.6/json/decoder.py", line 339, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/lib/python3.6/json/decoder.py", line 357, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
```

TCIII, 2021-11-28:
@beeplove,

Processor?
OS?
I am seeing python 3.6 while Rpi Buster is using 3.7.
Are you running a Nano? If so what JP version?

beeplove, 2021-12-03:
I was using the wrong version of TensorFlow.

closed 2021-12-03 by beeplove

Ezward, 2021-12-03:
@beeplove can we close this issue?

beeplove, 2021-12-04:
Yes, please