autorope/donkeycar #970: Installing imgaug on Jetson Nano break CSIC camera

Ezward, 2021-12-17:
This has been found using Jetpack 4.6.  In order to use the 'CROP' transformation with an autopilot the imgaug dependency needs to be installed into the environment.  It installs without issue.  however, once installed the CSIC camera will no longer return frames.  This is not an issue in RaspberryPi.  I'm not sure if it is an issue in earlier versions of Jetpack.  

Using 'CROP' is a best practice when training an autopilot.  So it is a problem that one of our supported platforms cannot use it. Worse, the error that is produced provides no help in understanding the underlying issue.  We should try to find a fix or work-around for this issue.

- Determine if earlier versions of jetpack also suffer from this issue
- Determine if there is an specific version of imgaug or one of it's dependencies that can be installed to avoid this issue.
- If we can't find a way to make imgaug work then open a issue for implementing CROP and trapezoidal masks that don't use imgaug.

Ezward, 2021-12-17:
See this related issue https://github.com/autorope/donkeycar/issues/749

Ezward, 2021-12-20:
when installing imgaug on the nano, I see a series of these errors while it is trying to install 
```
WARNING: Discarding https://files.pythonhosted.org/packages/44/ec/4eddbf9d17a917c51fb4ad159aa7137f506681e91ab559cf87d120e1d78d/Shapely-1.7.0.tar.gz#sha256=e21a9fe1a416463ff11ae037766fe410526c95700b9e545372475d2361cc951e (from https://pypi.org/simple/shapely/). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
  Using cached Shapely-1.6.4.post2.tar.gz (225 kB)
  Preparing metadata (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/ed/env/bin/python -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-dnuepv9f/shapely_3a52b345fc0746389fb9807cad9cd11a/setup.py'"'"'; __file__='"'"'/tmp/pip-install-dnuepv9f/shapely_3a52b345fc0746389fb9807cad9cd11a/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-204d_l62
       cwd: /tmp/pip-install-dnuepv9f/shapely_3a52b345fc0746389fb9807cad9cd11a/
  Complete output (11 lines):
  Failed `CDLL(libgeos_c.so.1)`
  Failed `CDLL(libgeos_c.so)`
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-install-dnuepv9f/shapely_3a52b345fc0746389fb9807cad9cd11a/setup.py", line 80, in <module>
      from shapely._buildcfg import geos_version_string, geos_version, \
    File "/tmp/pip-install-dnuepv9f/shapely_3a52b345fc0746389fb9807cad9cd11a/shapely/_buildcfg.py", line 167, in <module>
      fallbacks=['libgeos_c.so.1', 'libgeos_c.so'])
    File "/tmp/pip-install-dnuepv9f/shapely_3a52b345fc0746389fb9807cad9cd11a/shapely/_buildcfg.py", line 161, in load_dll
      libname, fallbacks or []))
  OSError: Could not find library geos_c or load any of its variants ['libgeos_c.so.1', 'libgeos_c.so']
  ----------------------------------------
```
I apperas to be looking for geos_c.so.  running this fixes those errors
```
sudo apt-get install libgeos-dev
```
However, it does not fix the camera issue, it still won't startup.

Ezward, 2021-12-20:
This is a simple way to reproduce the but that does not include all the donkeycar code.  

```
$ sudo apt-get install libgeos-dev
$ cd ~/projects
$ git clone https://github.com/JetsonHacksNano/CSI-Camera.git
$ cd ~/projects/CSI-Camera
$ rm -rf env
$ python3 -m virtualenv -p python3 env --system-site-packages
created virtual environment CPython3.6.9.final.0-64 in 1417ms
  creator CPython3Posix(dest=/home/ed/projects/CSI-Camera/env, clear=False, no_vcs_ignore=False, global=True)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/ed/.local/share/virtualenv)
    added seed packages: pip==21.3.1, setuptools==59.2.0, wheel==0.37.0
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
$ . env/bin/activate
$ python --version
Python 3.6.9

$ python simple_camera.py 
nvarguscamerasrc ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, format=(string)NV12, framerate=(fraction)60/1 ! nvvidconv flip-method=0 ! video/x-raw, width=(int)1280, height=(int)720, format=(string)BGRx ! videoconvert ! video/x-raw, format=(string)BGR ! appsink
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3264 x 2464 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3264 x 1848 FR = 28.000001 fps Duration = 35714284 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1640 x 1232 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 120.000005 fps Duration = 8333333 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 5 
   Output Stream W = 1280 H = 720 
   seconds to Run    = 0 
   Frame Rate = 120.000005 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
^CTraceback (most recent call last):
  File "simple_camera.py", line 66, in <module>
    show_camera()
  File "simple_camera.py", line 55, in show_camera
    keyCode = cv2.waitKey(30) & 0xFF
KeyboardInterrupt
GST_ARGUS: Cleaning up
CONSUMER: Done Success
GST_ARGUS: Done Success

$ pip install imgaug
Collecting imgaug
  Using cached imgaug-0.4.0-py2.py3-none-any.whl (948 kB)
Requirement already satisfied: six in /usr/local/lib/python3.6/dist-packages (from imgaug) (1.15.0)
Collecting opencv-python
  Using cached opencv_python-4.5.4.60-cp36-cp36m-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (39.0 MB)
Requirement already satisfied: scipy in /usr/lib/python3/dist-packages (from imgaug) (0.19.1)
Collecting imageio
  Using cached imageio-2.13.3-py3-none-any.whl (3.3 MB)
Collecting scikit-image>=0.14.2
  Using cached scikit_image-0.17.2-cp36-cp36m-linux_aarch64.whl
Collecting Shapely
  Using cached Shapely-1.8.0-cp36-cp36m-linux_aarch64.whl
Requirement already satisfied: numpy>=1.15 in /usr/local/lib/python3.6/dist-packages (from imgaug) (1.19.4)
Requirement already satisfied: matplotlib in /usr/lib/python3/dist-packages (from imgaug) (2.1.1)
Collecting Pillow
  Using cached Pillow-8.4.0-cp36-cp36m-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (3.0 MB)
Collecting tifffile>=2019.7.26
  Using cached tifffile-2020.9.3-py3-none-any.whl (148 kB)
Collecting PyWavelets>=1.1.1
  Using cached PyWavelets-1.1.1-cp36-cp36m-manylinux2014_aarch64.whl (4.3 MB)
Collecting networkx>=2.0
  Using cached networkx-2.5.1-py3-none-any.whl (1.6 MB)
Collecting scipy
  Using cached scipy-1.5.4-cp36-cp36m-manylinux2014_aarch64.whl (24.5 MB)
Collecting decorator<5,>=4.3
  Using cached decorator-4.4.2-py2.py3-none-any.whl (9.2 kB)
Installing collected packages: Pillow, decorator, tifffile, scipy, PyWavelets, networkx, imageio, Shapely, scikit-image, opencv-python, imgaug
  Attempting uninstall: decorator
    Found existing installation: decorator 4.1.2
    Not uninstalling decorator at /usr/lib/python3/dist-packages, outside environment /home/ed/projects/CSI-Camera/env
    Can't uninstall 'decorator'. No files were found to uninstall.
  Attempting uninstall: scipy
    Found existing installation: scipy 0.19.1
    Not uninstalling scipy at /usr/lib/python3/dist-packages, outside environment /home/ed/projects/CSI-Camera/env
    Can't uninstall 'scipy'. No files were found to uninstall.
Successfully installed Pillow-8.4.0 PyWavelets-1.1.1 Shapely-1.8.0 decorator-4.4.2 imageio-2.13.3 imgaug-0.4.0 networkx-2.5.1 opencv-python-4.5.4.60 scikit-image-0.17.2 scipy-1.5.4 tifffile-2020.9.3
(env) ed@diydonkeybot:~/projects/CSI-Camera$ python simple_camera.py 
nvarguscamerasrc ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, format=(string)NV12, framerate=(fraction)60/1 ! nvvidconv flip-method=0 ! video/x-raw, width=(int)1280, height=(int)720, format=(string)BGRx ! videoconvert ! video/x-raw, format=(string)BGR ! appsink
Unable to open camera
```

Ezward, 2021-12-20:
Here's a twist; I can reproduce the bug without installing imgaug.  Just install opencv-python:
```
$ cd ~/projects/CSI-Camera
$ rm -rf env
$ python3 -m virtualenv -p python3 env --system-site-packages
created virtual environment CPython3.6.9.final.0-64 in 1169ms
  creator CPython3Posix(dest=/home/ed/projects/CSI-Camera/env, clear=False, no_vcs_ignore=False, global=True)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/ed/.local/share/virtualenv)
    added seed packages: pip==21.3.1, setuptools==59.2.0, wheel==0.37.0
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
$ . env/bin/activate
$ python simple_camera.py 
nvarguscamerasrc ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, format=(string)NV12, framerate=(fraction)60/1 ! nvvidconv flip-method=0 ! video/x-raw, width=(int)1280, height=(int)720, format=(string)BGRx ! videoconvert ! video/x-raw, format=(string)BGR ! appsink
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3264 x 2464 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3264 x 1848 FR = 28.000001 fps Duration = 35714284 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1640 x 1232 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 120.000005 fps Duration = 8333333 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 5 
   Output Stream W = 1280 H = 720 
   seconds to Run    = 0 
   Frame Rate = 120.000005 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
^CTraceback (most recent call last):
  File "simple_camera.py", line 66, in <module>
    show_camera()
  File "simple_camera.py", line 55, in show_camera
    keyCode = cv2.waitKey(30) & 0xFF
KeyboardInterrupt
GST_ARGUS: Cleaning up
CONSUMER: Done Success
GST_ARGUS: Done Success

$ pip install opencv-python
Collecting opencv-python
  Using cached opencv_python-4.5.4.60-cp36-cp36m-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (39.0 MB)
Requirement already satisfied: numpy>=1.19.3 in /usr/local/lib/python3.6/dist-packages (from opencv-python) (1.19.4)
Installing collected packages: opencv-python
Successfully installed opencv-python-4.5.4.60

$ python simple_camera.py 
nvarguscamerasrc ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, format=(string)NV12, framerate=(fraction)60/1 ! nvvidconv flip-method=0 ! video/x-raw, width=(int)1280, height=(int)720, format=(string)BGRx ! videoconvert ! video/x-raw, format=(string)BGR ! appsink
Unable to open camera
```

Ezward, 2021-12-20:
I've been able to repeat this bug with the base Jetpack 4.6 python which is python3.6.9, and with python3.7.5 and python3.8.0

Ezward, 2021-12-20:
It works 'out-of-the-box' with no virtual environment, so this is what it is using for opencv:
```
ed@diydonkeybot:~/projects/CSI-Camera$ python3
Python 3.6.9 (default, Dec  8 2021, 21:08:43) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv2
>>> print(cv2)
<module 'cv2' from '/usr/lib/python3.6/dist-packages/cv2/python-3.6/cv2.cpython-36m-aarch64-linux-gnu.so'>
>>> print(cv2.__version__)
4.1.1
```

Ezward, 2021-12-20:
this works in the virtual environment is we don't try to install opencv-python
```
ed@diydonkeybot:~/projects/CSI-Camera$ rm -rf env
ed@diydonkeybot:~/projects/CSI-Camera$ python3 -m virtualenv -p python3 env --system-site-packages
created virtual environment CPython3.6.9.final.0-64 in 1291ms
  creator CPython3Posix(dest=/home/ed/projects/CSI-Camera/env, clear=False, no_vcs_ignore=False, global=True)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/ed/.local/share/virtualenv)
    added seed packages: pip==21.3.1, setuptools==59.2.0, wheel==0.37.0
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
ed@diydonkeybot:~/projects/CSI-Camera$ . env/bin/activate
(env) ed@diydonkeybot:~/projects/CSI-Camera$ python --version
Python 3.6.9
(env) ed@diydonkeybot:~/projects/CSI-Camera$ python simple_camera.py 
nvarguscamerasrc ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, format=(string)NV12, framerate=(fraction)60/1 ! nvvidconv flip-method=0 ! video/x-raw, width=(int)1280, height=(int)720, format=(string)BGRx ! videoconvert ! video/x-raw, format=(string)BGR ! appsink
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3264 x 2464 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3264 x 1848 FR = 28.000001 fps Duration = 35714284 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1640 x 1232 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 120.000005 fps Duration = 8333333 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 5 
   Output Stream W = 1280 H = 720 
   seconds to Run    = 0 
   Frame Rate = 120.000005 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
^CTraceback (most recent call last):
  File "simple_camera.py", line 66, in <module>
    show_camera()
  File "simple_camera.py", line 52, in show_camera
    ret_val, img = cap.read()
KeyboardInterrupt
GST_ARGUS: Cleaning up
CONSUMER: Done Success
GST_ARGUS: Done Success
(env) ed@diydonkeybot:~/projects/CSI-Camera$ python
Python 3.6.9 (default, Dec  8 2021, 21:08:43) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'cv'
>>> import cv2
>>> print(cv2.__version__)
4.1.1
```

Ezward, 2021-12-20:
Thanks @TCIII for finding this issue in the imgaug github, https://github.com/aleju/imgaug/issues/737

That indicated that the pip wheel install of imgaug will install opencv-python.  My own testing above shows this will conflict with the Jetpack opencv and will cause the camera error.

So I'm going to trying installing from source as indicated in that imgaug issue.
```
pip install imgaug --no-binary imgaug
```

Ezward, 2022-01-09:
The maintainer of the imgaug library has said they will not fix their installer.  So we should implement our augmentations another way.  There is an alternative library that used to use imgaug but has removed it because of this bug.  It may be a good replacement for us https://github.com/albumentations-team/albumentations#list-of-augmentations

Ezward, 2022-01-15:
[PR #978](https://github.com/autorope/donkeycar/pull/978) is a temporary workaround (read HACK) that mocks out the augmentations if imgaug dependency is not installed.  It is not installed by default on the Nano.  So this will prevent the camera issue.  It warns the user at runtime if they use a configuration that utilized augmentations.

## This temporary hack should be removed when we replace the imaug library with another implemenation.

naisy, 2022-01-25:
@Ezward You can uninstall opencv-python after installing imgaug 0.4.0

In my Dockerfile:
```
########################################
# imgaug 0.4.0
########################################
RUN  . ~/.virtualenv/$DONKEY_VERSION/bin/activate \
    && pip install six numpy scipy Pillow matplotlib scikit-image imageio Shapely \
    && pip install --no-dependencies imgaug \
    && pip install imagecorruptions \
    && pip uninstall -y opencv-python
```

Ezward, 2023-03-08:
see PR https://github.com/autorope/donkeycar/pull/1111

Ezward, 2023-03-22:
Replace imgaug with augmentations.  PR https://github.com/autorope/donkeycar/pull/1111

closed 2023-03-22 by Ezward