autorope/donkeycar #973: Cannot change modes from Web interface when using RC controller

zlite, 2021-12-18:
When using a RC controller (also when using a joystick?) the Web UI controls are disabled. Since the RC controller typically only has two channels that means that it's impossible to switch into Auto mode from the Web UI, since there is no spare channel to do that with.

Solution: enable Web UI controls (at least for modes) and start/stop when using RC control

Ezward, 2022-01-14:
Here is what I think the issue is.  The code first adds the web controller, then if configured, it will add in a game controller or RC controller.  See https://github.com/autorope/donkeycar/blob/a3df79a3573127dff31fc9b2953588056875b214/donkeycar/templates/complete.py#L198

The LocalWebController outputs `['user/angle', 'user/throttle', 'user/mode', 'recording']`.  So does the RC controller and they game controllers, but they do it last so they win.  Basically the RC/Game controllers overwrite whatever the web controller outputs.  So is appears the web controller is disabled, but it's not, it's output is just getting overwritten.

Ezward, 2022-01-15:
So for throttle and steering I think is is correct for the game/RC controller to overwrite values from the web controller because the user wants to use the physical inputs.  However, the user may want to use the web ui for other controls like user/mode, erase-records and arm-ai-launch.  

To handle this, we want all the controller to take those values and inputs only change them if the user makes a change using that controller, otherwise just pass through the values that it got.  That allows either controller to make a change and have it stick.

So the web controller, rc controller and joystick controllers should take `['user/mode', 'recording']` as inputs.  If that controller get's a user change for that value, then it should output the updated value otherwise it should just pass throught the value it got as an input.

zlite, 2022-01-15:
Exactly. You nailed it

sctse999, 2022-01-15:
I think I can work on this. This is something I want to change for quite a while.

Ezward, 2022-01-15:
I started on this; Code branch [970-enable-RC-controller-in-webui](https://github.com/autorope/donkeycar/tree/970-enable-RC-controller-in-webui)

Ezward, 2022-01-15:
Regarding making the webui reflect any changes made by the controller, there is a client websocket message handler in main.js, but it just logs the event.  So we need to send a websocket event to the javascript client with the state that has changed.  Then we need make the client's websocket message handler set the appropriate local state and call UpdateUI() to get it written to the DOM.  

This is a little harder than is sounds; we probably don't want the game controller code make websocket calls.  So we need some other part that is tracking the relevant states and when it sees the state change it will emit a websocket message to the web client.

Ezward, 2022-01-31:
This is the pull request to address this issue [PR #985](https://github.com/autorope/donkeycar/pull/985)

Ezward, 2022-02-02:
Addressed in https://github.com/autorope/donkeycar/pull/985

closed 2022-02-02 by Ezward