autorope/donkeycar #977: UnregisteredEnv: No registered env with id: donkey-generated-track-v0

jasperan, 2022-01-04:
Using a default install and following all steps from the tutorial in the documentation, the default track names to specify are:

# DONKEY_GYM_ENV_NAME = "donkey-generated-track-v0" # ("donkey-generated-track-v0"|"donkey-generated-roads-v0"|"donkey-warehouse-v0"|"donkey-avc-sparkfun-v0")

With the default value, which is donkey-generated-track-v0, I get the following error: (and any other name I put):

```
D:\github\donkey\mycar
(donkey) λ python manage.py drive
________             ______                   _________
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/
                                 /____/

using donkey v4.2.1 ...
loading config file: D:\github\donkey\mycar\config.py
loading personal config over-rides from myconfig.py
INFO:root:PID: 14584
INFO:root:cfg.CAMERA_TYPE PICAM
cfg.CAMERA_TYPE PICAM
Traceback (most recent call last):
  File "D:\Programs\miniconda3\envs\donkey\lib\site-packages\gym\envs\registration.py", line 158, in spec
    return self.env_specs[id]
KeyError: 'donkey-generated-track-v0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "manage.py", line 719, in <module>
    meta=args['--meta'])
  File "manage.py", line 135, in drive
    cam = DonkeyGymEnv(cfg.DONKEY_SIM_PATH, host=cfg.SIM_HOST, env_name=cfg.DONKEY_GYM_ENV_NAME, conf=cfg.GYM_CONF, record_location=cfg.SIM_RECORD_LOCATION, record_gyroaccel=cfg.SIM_RECORD_GYROACCEL, record_velocity=cfg.SIM_RECORD_VELOCITY, delay=cfg.SIM_ARTIFICIAL_LATENCY)
  File "d:\github\donkey\donkeycar\donkeycar\parts\dgym.py", line 25, in __init__
    self.env = gym.make(env_name, conf=conf)
  File "D:\Programs\miniconda3\envs\donkey\lib\site-packages\gym\envs\registration.py", line 235, in make
    return registry.make(id, **kwargs)
  File "D:\Programs\miniconda3\envs\donkey\lib\site-packages\gym\envs\registration.py", line 128, in make
    spec = self.spec(path)
  File "D:\Programs\miniconda3\envs\donkey\lib\site-packages\gym\envs\registration.py", line 203, in spec
    raise error.UnregisteredEnv("No registered env with id: {}".format(id))
gym.error.UnregisteredEnv: No registered env with id: donkey-generated-track-v0
```

Bleyddyn, 2022-01-20:
That should work. I just tested it with the latest donkeycar and gym_donkeycar repos.

Donkey v4.3.4 and gym-donkeycar v1.1.1

Ezward, 2022-02-15:
Here is a teacher that has a number of students that got the same issue.  https://discord.com/channels/662098530411741184/671604071999864832/943017438193279006. Using donkey 4.2.1 like the original poster.

s2718, 2022-02-15:
> That should work. I just tested it with the latest donkeycar and gym_donkeycar repos.
> 
> Donkey v4.3.4 and gym-donkeycar v1.1.1

Where do you find that version of Donkey?