autorope/donkeycar #984: The Jetson Nano will be able to provide the necessary GPIO pin support?

Anton890304, 2022-01-20:
The official instructions for the latest donkey car release say - The Jetson Nano does not provide the necessary GPIO pin support. https://docs.donkeycar.com/parts/rc/

On the Jetson community page, I saw a Jetson.GPIO Python library, that promises support API as RPi.GPIO. https://forums.developer.nvidia.com/t/pigpio-support-on-nano/74744?ncid=afm-chs-44270&ranMID=44270&ranEAID=a1LgFw09t88&ranSiteID=a1LgFw09t88-dDM01oFlH6O02ohKT51p0A

Does this mean that there will soon be a donkey car - dev. version of the where we can connect the RC receiver directly to the GPIO on Jetson Nano?

adricl, 2022-01-26:
Donkeycar on the jetson nano talks to the PCA9685 via the i2c gpio pins. You don't need the gpio interface as the jetson exposes these pins via the Linux i2c kernel interface.

Ezward, 2022-02-07:
Output on the Jetson GPIO works.  PWM works if you enable hardware PWM, so you can drive motors; see the docs https://docs.donkeycar.com/parts/pins/#generating-pwm-from-the-jetson-nano  Input, like reading encoders, does not work well; in that case use an arduino to read the encoders.  This branch implements that; https://github.com/autorope/donkeycar/tree/921-next-generation-odometer-parts  We hope to have the merges in a couple of weeks (at Feb 19 2022 race if we can).

Ezward, 2022-02-07:
@Anton890304 please let us know if this answers your question.  We would like to close this issue.

closed 2022-02-07 by Anton890304

Ezward, 2022-02-12:
A little more background on the jetson GPIO issue.  TLDR; fix requires a pull request in Jetson.GPIO library.

Jetson.GPIO
- Jetson.GPIO works for well for output pins. 
- For PWM pins you must go through the process on enabling hardward PWM, then you can get PWM signals on two pins.
- Input pins work fine in polled mode.
- Input pins work fine in interrupt mode IFF your are only running on Python's main thread (so no background threads) AND if your callback function never throw and exception. I have tested interrupt driven input pins successfully in several contexts;
  - In ROS (specifically on a Duckietown DuckieBot) I have successfully read two mono-encoders using interrupts.  The ROS node is running it's own process and so the Python code is not threaded; it is dedicated to just publishing the ticks it gets based on the interrupt callbacks.
  - In Donkeycar the pins.py self test can be configured to use an interrupt to read a pin.  I have attached this to the input of high resolution (350 ppr) wheel encoders successfully.  In this case the self test is not threaded.
  - In Donkeycar the tachometer.py self test can be configured to use an encoder that reads a GPIO using the pins.py api and an interrupt driven input pin.  This works well.  In this case the self test is not thread.
- Input pins in interrupt mode do not work at all in the general Donkeycar framework.  Specifically I cannot get tachometer.py to work, even though the self test works.  What happens is that it will get one callback successfully then never get any other callbacks.  I believe this is because of all the other threaded parts.  Having read the code, I think there are bugs in the Jetson.GPIO that cause the thread that reads the GPIO bail out if 1) the callback causes any exception 2) it gets a thread interrupt at an inopportune time. In either of these cases the thread will get killed an no callbacks will get called anymore.  So I think this happens in the donkey environment because we have a lot of background threads and so the GPIO thread is receiving thread interrupts, which it is not handling gracefully.  The Jetsion.GPIO thread is basically a state machine and if it get's interrupted at the wrong time the state get's messed up.  
- No one seems to be working on Jetson.GPIO at the moment.  I think it would not take much effort to protect the GPIO thread so it worked in the donkey environment.  That would require a pull request.  
- For now I added an Arduino Nano to my robot and use it to read the quadrature encoders and it works great.

Anton890304, 2022-02-13:
@Ezward Thank you very much for your detailed answer!