autorope/donkeycar #988: donkey: command not found error after install on mac

c1505, 2022-02-06:
I followed the instructions here https://docs.donkeycar.com/guide/host_pc/setup_mac/ to install on a host PC .  Specifically using the environment setup here and not the mamba one
```
conda env create -f install/envs/mac.yml
conda activate donkey
pip install -e .[pc]
```
I didn't see any errors in the install, but I get an error when trying to use the software
```
(donkey) coreys-mbp:donkeycar corey$ donkey createcar --path ~/mycar
bash: donkey: command not found
```
Where should I expect the `donkey` software to be installed ? Maybe I am missing something from my path ? 
Or I would take any other advice for how to troubleshoot this or solutions.  
Thanks.

Ezward, 2022-02-06:
Try running the donkey command from within the donkeycar project folder that you cloned from github.

c1505, 2022-02-06:
Tried that. Still the same.  What is the executable that I am trying to target with that command ? 
I also looked where I would expect to find the donkeycar file in my path $HOME/opt/anaconda3/envs/donkey/bin .  Nothing named donkeycar there.  Assuming the conda environment is named donkey is that where I would expect it to be ?

DocGarbanzo, 2022-02-06:
@c1505 - the `donkey` command should be available from the `donkey` environment. This is defined as entry point in `setup.py`. If it's not working for you, it means your install has gone wrong somehow. Can you show the output of your `pip install -e .[pc]`?

c1505, 2022-02-07:
```
(donkey) coreys-mbp:donkeycar corey$ pip install -e .[pc]
DEPRECATION: Configuring installation scheme with distutils config files is deprecated and will no longer work in the near future. If you are using a Homebrew or Linuxbrew Python, please see discussion at https://github.com/Homebrew/homebrew-core/issues/76621
Obtaining file:///Users/corey/projects/donkeycar
  Preparing metadata (setup.py) ... done
Collecting numpy
  Downloading numpy-1.22.2-cp39-cp39-macosx_10_14_x86_64.whl (17.6 MB)
     |████████████████████████████████| 17.6 MB 9.4 MB/s            
Requirement already satisfied: pillow in /usr/local/lib/python3.9/site-packages (from donkeycar==4.2.1) (8.4.0)
Collecting docopt
  Downloading docopt-0.6.2.tar.gz (25 kB)
  Preparing metadata (setup.py) ... done
Collecting tornado
  Downloading tornado-6.1-cp39-cp39-macosx_10_9_x86_64.whl (416 kB)
     |████████████████████████████████| 416 kB 4.7 MB/s            
Requirement already satisfied: requests in /usr/local/lib/python3.9/site-packages (from donkeycar==4.2.1) (2.26.0)
Collecting h5py
  Downloading h5py-3.6.0-cp39-cp39-macosx_10_9_x86_64.whl (3.1 MB)
     |████████████████████████████████| 3.1 MB 4.0 MB/s            
Collecting PrettyTable
  Downloading prettytable-3.0.0-py3-none-any.whl (24 kB)
Collecting paho-mqtt
  Downloading paho-mqtt-1.6.1.tar.gz (99 kB)
     |████████████████████████████████| 99 kB 12.9 MB/s            
  Preparing metadata (setup.py) ... done
Collecting simple_pid
  Using cached simple_pid-1.0.1-py2.py3-none-any.whl (8.1 kB)
Collecting progress
  Downloading progress-1.6.tar.gz (7.8 kB)
  Preparing metadata (setup.py) ... done
Collecting typing_extensions
  Downloading typing_extensions-4.0.1-py3-none-any.whl (22 kB)
Collecting pyfiglet
  Downloading pyfiglet-0.8.post1-py2.py3-none-any.whl (865 kB)
     |████████████████████████████████| 865 kB 10.9 MB/s            
Collecting psutil
  Downloading psutil-5.9.0-cp39-cp39-macosx_10_9_x86_64.whl (238 kB)
     |████████████████████████████████| 238 kB 12.3 MB/s            
Collecting matplotlib
  Downloading matplotlib-3.5.1-cp39-cp39-macosx_10_9_x86_64.whl (7.3 MB)
     |████████████████████████████████| 7.3 MB 12.5 MB/s            
Collecting imgaug
  Downloading imgaug-0.4.0-py2.py3-none-any.whl (948 kB)
     |████████████████████████████████| 948 kB 19.4 MB/s            
Collecting kivy
  Downloading Kivy-2.0.0-cp39-cp39-macosx_10_14_x86_64.macosx_10_9_intel.macosx_10_9_x86_64.macosx_10_10_intel.macosx_10_10_x86_64.whl (7.2 MB)
     |████████████████████████████████| 7.2 MB 18.5 MB/s            
Collecting Shapely
  Downloading Shapely-1.8.0-cp39-cp39-macosx_10_9_x86_64.whl (1.2 MB)
     |████████████████████████████████| 1.2 MB 20.3 MB/s            
Collecting scikit-image>=0.14.2
  Downloading scikit_image-0.19.1-cp39-cp39-macosx_10_13_x86_64.whl (13.2 MB)
     |████████████████████████████████| 13.2 MB 29.1 MB/s            
Collecting scipy
  Downloading scipy-1.8.0-cp39-cp39-macosx_12_0_universal2.macosx_10_9_x86_64.whl (55.6 MB)
     |████████████████████████████████| 55.6 MB 12.9 MB/s            
Requirement already satisfied: six in /usr/local/lib/python3.9/site-packages (from imgaug->donkeycar==4.2.1) (1.15.0)
Collecting imageio
  Downloading imageio-2.14.1-py3-none-any.whl (3.3 MB)
     |████████████████████████████████| 3.3 MB 27.3 MB/s            
Collecting opencv-python
  Downloading opencv_python-4.5.5.62-cp36-abi3-macosx_10_15_x86_64.whl (46.0 MB)
     |████████████████████████████████| 46.0 MB 11.3 MB/s            
Collecting pygments
  Downloading Pygments-2.11.2-py3-none-any.whl (1.1 MB)
     |████████████████████████████████| 1.1 MB 7.8 MB/s            
Collecting Kivy-Garden>=0.1.4
  Downloading kivy-garden-0.1.4.tar.gz (6.8 kB)
  Preparing metadata (setup.py) ... done
Collecting docutils
[... 57 lines cut ...]
  DEPRECATION: Configuring installation scheme with distutils config files is deprecated and will no longer work in the near future. If you are using a Homebrew or Linuxbrew Python, please see discussion at https://github.com/Homebrew/homebrew-core/issues/76621
  DEPRECATION: Configuring installation scheme with distutils config files is deprecated and will no longer work in the near future. If you are using a Homebrew or Linuxbrew Python, please see discussion at https://github.com/Homebrew/homebrew-core/issues/76621
  DEPRECATION: Configuring installation scheme with distutils config files is deprecated and will no longer work in the near future. If you are using a Homebrew or Linuxbrew Python, please see discussion at https://github.com/Homebrew/homebrew-core/issues/76621
  DEPRECATION: Configuring installation scheme with distutils config files is deprecated and will no longer work in the near future. If you are using a Homebrew or Linuxbrew Python, please see discussion at https://github.com/Homebrew/homebrew-core/issues/76621
  DEPRECATION: Configuring installation scheme with distutils config files is deprecated and will no longer work in the near future. If you are using a Homebrew or Linuxbrew Python, please see discussion at https://github.com/Homebrew/homebrew-core/issues/76621
  DEPRECATION: Configuring installation scheme with distutils config files is deprecated and will no longer work in the near future. If you are using a Homebrew or Linuxbrew Python, please see discussion at https://github.com/Homebrew/homebrew-core/issues/76621
[... cut ...] gress-1.6 psutil-5.9.0 pyfiglet-0.8.post1 pygments-2.11.2 python-dateutil-2.8.2 scikit-image-0.19.1 scipy-1.8.0 simple-pid-1.0.1 tifffile-2022.2.2 tornado-6.1 typing-extensions-4.0.1 wcwidth-0.2.5
WARNING: You are using pip version 21.3.1; however, version 22.0.3 is available.
You should consider upgrading via the '/usr/local/opt/python@3.9/bin/python3.9 -m pip install --upgrade pip' command.
(donkey) coreys-mbp:donkeycar corey$ donkey createcar --path ~/mycar
bash: donkey: command not found
```

c1505, 2022-02-07:
@DocGarbanzo I have a linux PC that appeared to install successfully so this is not a problem for me at the moment.  I am not sure if it will become one in the future as my linux machine is a desktop and I may want to do some things outside the home.

DocGarbanzo, 2022-02-18:
@c1505 - please install donkey through conda  from the yaml file, as explained in the docs. Then you run pip in the donkey directory, as in the docs. You seem to have installed from pip directly and that created a python 3.9 environment, but you need to run python 3.7 for donkey. We'll be upgrading to python 3.9 / tf 2.6 soon hopefully.

closed 2022-02-18 by DocGarbanzo

c1505, 2022-02-20:
Thanks.  I did install the donkey through the conda in the yaml file.  Found out what my problem was.  **Not a donkeycar issue.  So feel free to ignore the following unless you are just curious**.  There is something funky going on with the terminal startup and environment when I use the vscode terminal.  It appears to show as starting in the `base` conda environment , but isn't using the python3 version from that environment and appears to be using the system python.  If I activate the donkey environment without deactivating this `base` environment first, it still tries to use that incorrect python version.  I have to deactivate the base environment and reactivate it in order for it to show the correct python3 version for whatever environment in question.  That is probably very confusing.  Seeing the terminal output likely will make more sense.  
```
The default interactive shell is now zsh.
To update your account to use zsh, please run `chsh -s /bin/zsh`.
For more details, please visit https://support.apple.com/kb/HT208050.
(base) coreys-mbp:~ corey$ python3 --version
Python 3.9.10
(base) coreys-mbp:~ corey$ conda env list
# conda environments:
#
base                  *  /Users/corey/opt/anaconda3
donkey                   /Users/corey/opt/anaconda3/envs/donkey
myenv                    /Users/corey/opt/anaconda3/envs/myenv
myenv2                   /Users/corey/opt/anaconda3/envs/myenv2
p37                      /Users/corey/opt/anaconda3/envs/p37
self-driving             /Users/corey/opt/anaconda3/envs/self-driving

(base) coreys-mbp:~ corey$ conda activate donkey
(donkey) coreys-mbp:~ corey$ conda env list
# conda environments:
#
base                     /Users/corey/opt/anaconda3
donkey                *  /Users/corey/opt/anaconda3/envs/donkey
myenv                    /Users/corey/opt/anaconda3/envs/myenv
myenv2                   /Users/corey/opt/anaconda3/envs/myenv2
p37                      /Users/corey/opt/anaconda3/envs/p37
self-driving             /Users/corey/opt/anaconda3/envs/self-driving

(donkey) coreys-mbp:~ corey$ python3 --version
Python 3.9.10
(donkey) coreys-mbp:~ corey$ conda deactivate
(base) coreys-mbp:~ corey$ python3 --version
Python 3.9.10
(base) coreys-mbp:~ corey$ conda deactivate
coreys-mbp:~ corey$ python3 --version
Python 3.9.10
coreys-mbp:~ corey$ conda activate base
(base) coreys-mbp:~ corey$ python3 --version
Python 3.8.5
(base) coreys-mbp:~ corey$ conda activate donkey
(donkey) coreys-mbp:~ corey$ python3 --version
Python 3.7.11
```

DocGarbanzo, 2022-02-23:
So you seem to have two different donkey envs,  you should be able to identify them by calling `which python` after you activated them like above. Maybe also call `conda info` in the different envs to find any clues.