autorope/donkeycar #989: Steering does not work when using drive , but does in calibration .  Earlier commit works .

c1505, 2022-02-07:
#### Issue description
Steering does not work when using drive , but does in calibration and it works when using an earlier commit.  Seeing that another person in discord reported the same problem and that [downgrading to 4.2.1 fixed it](https://discord.com/channels/662098530411741184/671604048218292233/935889345460699176) , I tried a more recent commit `d5b91301ff904873e70b9caba72823fe3454feb6`.  It fixed the problem.  I am using a Jetson nano 2gb and the other person has a raspberry pi so it seems to not be related to the particular board used.  

#### Steps to reproduce the issue

1.  Follow software setup as described here for the [jetson nano ](https://docs.donkeycar.com/guide/robot_sbc/setup_jetson_nano/)
2. create donkeycar from template
3. perform calibration and modify myconfig.py
4. changes to myconfig.py for my car
```
PCA9685_I2C_BUSNUM = 1
CAMERA_TYPE = "CSIC"   # (PICAM|WEBCAM|CVCAM|CSIC|V4L|D435|MOCK|IMAGE_LIST)
IMAGE_W = 224
IMAGE_H = 224
PCA9685_I2C_BUSNUM = 1
STEERING_LEFT_PWM = 460         #pwm value for full left steering
STEERING_RIGHT_PWM = 260        #pwm value for full right steering
THROTTLE_FORWARD_PWM = 420      #pwm value for max forward throttle
THROTTLE_STOPPED_PWM = 370      #pwm value for no movement
THROTTLE_REVERSE_PWM = 320      #pwm value for max reverse throttle
``` 
5. initiate driving from the car folder `python manage.py drive`
6. navigate to browser
7. press j multiple times to try to turn wheels left
8. press k multiple times to try to turn wheels right
9. also tested on mobile with the touch interface

#### What's the expected result?

- wheels turn

#### What's the actual result?

- wheels do not turn

#### Additional details / screenshot

- the webrowser UI shows changes to the steering, but no changes happen in the car.  The throttle still works. 
- the working commit `d5b91301ff904873e70b9caba72823fe3454feb6` is from December 30th of 2021.  I have not tested other commits yet.  I figured that someone more familiar with the code would be able to narrow down where the problem originated better than me.

Ezward, 2022-02-07:
So that change got made to docs because we almost always tell folks to use dev when they have a problem.  However, I believe you have found a pretty big bug in dev; we have duplicated this in another branch that branched from dev.  As a work around, I think the I2C_SERVO drivetrain will work; it worked in that other branch.  Thanks for reporting this bug.

Ezward, 2022-02-07:
I'm going to revert that change to docs because it will lead folks to a bad experience.

c1505, 2022-02-08:
Thanks for the fast response @Ezward .  This is a really cool project!  I appreciate the work you all are doing improving it and keeping it going 🥇

TCIII, 2022-02-08:
@c1505

I am a DC Maintainer and have duplicated your issue using DRIVE_TRAIN_TYPE = "PWM_STEERING_THROTTLE" with both a gamepad (PS4) and the webui when running the 'path_follow.py' template.
I have throttle control, but no steering control.
If I set DRIVE_TRAIN_TYPE = "I2C_SERVO" I have throttle and steering control with both the gamepad and the webui.

TCIII

Ezward, 2022-02-11:
I believe what is going on here is that the PWM_STEERING_PIN configuration values is used in several DRIVETRAIN_TYPE configurations, each with a different default value.  If you just set the DRIVETRAIN_TYPE = "PWM_STEERING_THROTTLE", but don't uncomment the PWM_STEERING_PIN value, then the default of a different drivetrain get's used.  

So a workaround is to just uncomment PWM_STEERING_PIN in the PWM_STEERING_THROTTLE sections.

I will create a real fix and make a pull request for dev.

Ezward, 2022-03-02:
Proposed fix is in this branch.  https://github.com/autorope/donkeycar/tree/989-namespace-drivetrain-configurations  @c1505 Corey if you get a change to try this it would be appreciated.

c1505, 2022-03-02:
@Ezward Thanks !  I'll give it a try today or tomorrow. Want to open a draft PR as well to have the automated tests kick off and to make the changes more easily visible ?

c1505, 2022-03-05:
Sorry I had a busy week and am away this weekend. I will give it a shot Monday.  Thanks again for the fix!

BrianHenryIE, 2022-03-07:
I tried:

```
git checkout -b 989-namespace-drivetrain-configurations origin/989-namespace-drivetrain-configurations
```

with no other changes and it did not work.

> just uncomment PWM_STEERING_PIN in the PWM_STEERING_THROTTLE sections.

That worked. Thank you.

Ezward, 2022-03-07:
> I tried:
> 
> ```
> git checkout -b 989-namespace-drivetrain-configurations origin/989-namespace-drivetrain-configurations
> ```
> 
> with no other changes and it did not work.
> 
@BrianHenryIE can you give me a little more info on this; what did you do?  Did you create a new mycar?  Did you update your prior mycar using `donkey createcar --overwrite --path <yourcar>`?  Did it fail the exact same way or a different way?
Thanks

BrianHenryIE, 2022-03-07:
I did not create a new mycar, nor did I update. (my existing mycar was vanilla – I had reinstalled from scratch as I tried to troubleshoot).

The fail was in the same way, i.e. no error message, just no working steering when launched.

Ezward, 2022-03-07:
> I did not create a new mycar, nor did I update. (my existing mycar was vanilla – I had reinstalled from scratch as I tried to troubleshoot).
> 
> The fail was in the same way, i.e. no error message, just no working steering when launched.

Ok, then it failed because there was no change to myconfig.py; so it failed in the same way.  The issue is with how myconfig.py was being generated (with duplicate values in different sections).  Thank you for the details.

Ezward, 2022-06-17:
Yes, need to update docs to instruct user to uncomment the entire dictionary block for the chosen drivetrain.  What would be better is that they are just uncommented to start, since they do not conflict.  Then setting the DRIVETRAIN_TYPE will just choose the right one.  I'll look and see if we can do that.

closed 2022-10-12 by Ezward