autorope/donkeycar #996: JPG files don't recording in the data folder

Anton890304, 2022-03-09:
I found the issue in Donkey Car version 4.3.5 from the dev branch. Dataset writing doesn't work. JPG files don't recording in the data folder.
In the Donkey Car web interface, the (Start Recording [r]) button doesn't  switch to recording mode. At that time in the JupyterLab terminal I see only: `INFO:donkeycar.parts.web_controller.web:{'recording': False}`
When toggling recording on/off on the gamepad in the JupyterLab terminal, I see only: `recording: False`

- Hardware - JetsonNano
- myconfig.py parameter AUTO_RECORD_ON_THROTTLE = True

DocGarbanzo, 2022-03-15:
Hi @Anton890304, are you also driving? Because you will only record images when you use throttle. Also, what do you mean by JupyterLab terminal?

Anton890304, 2022-03-16:
> Привет @Anton890304 , ты тоже за рулем? Потому что вы будете записывать изображения только при использовании газа. Кроме того, что вы подразумеваете под терминалом JupyterLab?

I push the throttle, but recording doesn't work.
Yes i use JupyterLab + web interface - :8887/drive

Anton890304, 2022-03-16:
@DocGarbanzo 
I push the throttle, but recording doesn't work.
Yes i use JupyterLab + web interface - :8887/drive

BrianHenryIE, 2022-03-20:
I have the same issue on a fresh Donkey install on Jetson Nano (not a fresh Ubuntu install).

Same problem existed with PS4 controller, but this output is web controller. The only changes in `myconfig.py` are the camera and `PCA9685_I2C_BUSNUM = 1`.

```
(env) brian@briannano:~/projects/donkeycar$ python ~/mycar/manage.py drive 
________             ______                   _________              
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /    
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/     
                                 /____/                              

using donkey v4.3.7 ...
WARNING:donkeycar.parts.pins:pigpio was not imported.
loading config file: /home/brian/mycar/config.py
loading personal config over-rides from myconfig.py
INFO:__main__:PID: 9427
cfg.CAMERA_TYPE CSIC
INFO:__main__:cfg.CAMERA_TYPE CSIC
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3264 x 2464 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3264 x 1848 FR = 28.000001 fps Duration = 35714284 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1640 x 1232 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 120.000005 fps Duration = 8333333 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 5 
   Output Stream W = 1280 H = 720 
   seconds to Run    = 0 
   Frame Rate = 120.000005 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
[ WARN:0] global /home/nvidia/host/build_opencv/nv_opencv/modules/videoio/src/cap_gstreamer.cpp (933) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
INFO:donkeycar.parts.camera:CSICamera opened...
INFO:donkeycar.parts.camera:...warming camera
INFO:donkeycar.parts.camera:CSICamera ready.
INFO:donkeycar.vehicle:Adding part CSICamera.
Starting Donkey Server...... you can now go to briannano.local:8887 to drive your car.
INFO:donkeycar.vehicle:Adding part LocalWebController.
INFO:donkeycar.vehicle:Adding part ThrottleFilter.
INFO:donkeycar.vehicle:Adding part PilotCondition.
INFO:donkeycar.vehicle:Adding part RecordTracker.
INFO:donkeycar.vehicle:Adding part DriveMode.
INFO:donkeycar.vehicle:Adding part AiLaunch.
INFO:donkeycar.vehicle:Adding part AiRunCondition.
INFO:donkeycar.parts.actuator:PWM Steering created
INFO:donkeycar.parts.actuator:Init ESC
INFO:donkeycar.parts.actuator:PWM Throttle created
INFO:donkeycar.vehicle:Adding part PWMSteering.
INFO:donkeycar.vehicle:Adding part PWMThrottle.
Using catalog /home/brian/mycar/data/catalog_0.catalog
INFO:donkeycar.vehicle:Adding part TubWriter.
You can now go to <your hostname.local>:8887 to drive your car.
INFO:donkeycar.vehicle:Starting vehicle at 20 Hz
INFO:tornado.access:304 GET /drive (fe80::c75:5368:eb1a:3d6e%wlan0) 15.38ms
INFO:tornado.access:200 GET /static/jquery-3.1.1.min.js (fe80::c75:5368:eb1a:3d6e%wlan0) 17.98ms
INFO:tornado.access:200 GET /static/nipple.js (fe80::c75:5368:eb1a:3d6e%wlan0) 12.91ms
INFO:tornado.access:200 GET /static/bootstrap/3.3.7/js/bootstrap.min.js (fe80::c75:5368:eb1a:3d6e%wlan0) 13.96ms
[... 20 lines cut ...]
|      AiLaunch      | 0.08 | 0.01 | 0.02 | 0.02 | 0.02 | 0.04 |  0.06 |
|   AiRunCondition   | 0.09 | 0.01 | 0.01 | 0.01 | 0.02 | 0.02 |  0.07 |
|    PWMSteering     | 0.11 | 0.02 | 0.03 | 0.03 | 0.04 | 0.06 |  0.10 |
|    PWMThrottle     | 0.07 | 0.01 | 0.02 | 0.02 | 0.03 | 0.04 |  0.07 |
+--------------------+------+------+------+------+------+------+-------+
(env) brian@briannano:~/projects/donkeycar$ ls -l ~/mycar/data
total 12
-rw-rw-r-- 1 brian brian    0 Mar 20 15:15 catalog_0.catalog
-rw-rw-r-- 1 brian brian  111 Mar 20 15:15 catalog_0.catalog_manifest
drwxrwxr-x 2 brian brian 4096 Mar 20 15:15 images
-rw-rw-r-- 1 brian brian  236 Mar 20 15:15 manifest.json
(env) brian@briannano:~/projects/donkeycar$ ls -l ~/mycar/data/images
total 0
```

The "start recording" button in the web interface never changed (brief flash) when I clicked it:

[image]

And when I clicked the button on the PS4 controller in an earlier attempt, this was the output:

```
auto record on throttle is enabled.
recording: False
```

Anton890304, 2022-03-21:
@BrianHenryIE
Did you install the Donkey Car following the instructions from [docs.donkeycar.com](https://docs.donkeycar.com/guide/robot_sbc/setup_jetson_nano/) or the instructions from [https://github.com/autorope/donkeycar/blob/main/install/nano/install-jp44.sh](https://github.com/autorope/donkeycar/blob/main/install/nano/install-jp44.sh) ?

I installed Donkey Cart following the instructions https://github.com/autorope/donkeycar/blob/main/install/nano/install-jp44.sh

[image]

Consequently. I have installed the recommended versions, but the Recording does not work.

**What do you think is the problem?**

BrianHenryIE, 2022-03-21:
The `/jp/v44` on that install .sh file is older than the docs which uses `jp/v45`, so I did not follow that.

For the most part, I did follow docs.donkeycar.com. But, as I said above, I'm trying to add a new part with higher requirements so I did not follow them strictly.

If I were you, I would follow docs.donkeycar.com instructions. It sounds like neither of use followed them properly, so it would be a good way to narrow down was it our own fault.

Best yet, if we export our broken environment with:

```
pip freeze > issue996.txt
```

Then once it is working we can figure what package difference caused the issue.

Ezward, 2022-03-23:
It's possible it is a bug that got added with this pull request https://github.com/autorope/donkeycar/pull/985  That PR was to make the webui and the game controllers play nice together.  There was a lot of state passed around that had to be synchonized; it's possible I messed up the recording state.

Ezward, 2022-03-23:
I think the issue may be with that PR;  line 258 of donkeycar/parts/web_controller/web.py;    [ self.application.recording = data.get('recording', self.application.recording) ](https://github.com/autorope/donkeycar/blob/58979613c723aadf6b2fa98b4a43743e18ab1641/donkeycar/parts/web_controller/web.py#L258)  should probably be 
```
        if data.get('recording') is not None:
            self.application.recording = data['recording']
            self.application.recording_latch = self.application.recording
```

That value is not getting latched, and so it getting overwritten.  @BrianHenryIE  or @Anton890304 Please try making that change and see if it works for you.

Ezward, 2022-04-05:
There is a duplicate of this issue here: https://github.com/autorope/donkeycar/issues/1005

Anton890304, 2022-04-06:
@Ezward Thank you very much))))

Ezward, 2022-04-07:
The fix is employed in this PR https://github.com/autorope/donkeycar/pull/1007 which is now merged.

closed 2022-04-07 by Ezward