autorope/donkeycar #999: Can't record data on ubuntu 20.04

Philan014, 2022-03-22:
on the 4.3.6 I get this info when i drive:

INFO:donkeycar.parts.web_controller.web:{'driveMode': 'user'}

and this info if i try to record data:
INFO:donkeycar.parts.web_controller.web:{'recording': False}

on the 4.3.7 i dont get any info messages but it also doesnt record the data.

I tried using only web controller for recording manually as well as record on throttle. Also tried using ps4 controller to record the data in both ways and it doesnt record either.

the only changed part in myconfig.py when using web controller:

DONKEY_GYM = True
DONKEY_SIM_PATH = "/home/gutek/projects/DonkeySimLinux/donkey_sim.x86_64"

the config.py has not been changed and looks as the following:

"""
CAR CONFIG

This file is read by your car application's manage.py script to change the car
performance.

EXAMPLE
-----------
import dk
cfg = dk.load_config(config_path='~/mycar/config.py')
print(cfg.CAMERA_RESOLUTION)

"""

import os

#PATHS
CAR_PATH = PACKAGE_PATH = os.path.dirname(os.path.realpath(__file__))
DATA_PATH = os.path.join(CAR_PATH, 'data')
MODELS_PATH = os.path.join(CAR_PATH, 'models')

#VEHICLE
DRIVE_LOOP_HZ = 20      # the vehicle loop will pause if faster than this speed.
MAX_LOOPS = None        # the vehicle loop can abort after this many iterations, when given a positive integer.

#CAMERA
CAMERA_TYPE = "PICAM"   # (PICAM|WEBCAM|CVCAM|CSIC|V4L|D435|MOCK|IMAGE_LIST)
IMAGE_W = 160
IMAGE_H = 120
IMAGE_DEPTH = 3         # default RGB=3, make 1 for mono
CAMERA_FRAMERATE = DRIVE_LOOP_HZ
CAMERA_VFLIP = False
CAMERA_HFLIP = False
CAMERA_INDEX = 0  # used for 'WEBCAM' and 'CVCAM' when there is more than one camera connected 
# For CSIC camera - If the camera is mounted in a rotated position, changing the below parameter will correct the output frame orientation
CSIC_CAM_GSTREAMER_FLIP_PARM = 0 # (0 => none , 4 => Flip horizontally, 6 => Flip vertically)

# For IMAGE_LIST camera
# PATH_MASK = "~/mycar/data/tub_1_20-03-12/*.jpg"

#9865, over rides only if needed, ie. TX2..
PCA9685_I2C_ADDR = 0x40     #I2C address, use i2cdetect to validate this number
PCA9685_I2C_BUSNUM = None   #None will auto detect, which is fine on the pi. But other platforms should specify the bus num.

#SSD1306_128_32
USE_SSD1306_128_32 = False    # Enable the SSD_1306 OLED Display
SSD1306_128_32_I2C_ROTATION = 0 # 0 = text is right-side up, 1 = rotated 90 degrees clockwise, 2 = 180 degrees (flipped), 3 = 270 degrees
SSD1306_RESOLUTION = 1 # 1 = 128x32; 2 = 128x64

#
# DRIVE_TRAIN_TYPE
# These options specify which chasis and motor setup you are using.
[... 483 lines cut ...]
PATH_OFFSET = (0, 0)                # 255, 255 is the center of the map. This offset controls where the origin is displayed.
PATH_MIN_DIST = 0.3                 # after travelling this distance (m), save a path point
PID_P = -10.0                       # proportional mult for PID path follower
PID_I = 0.000                       # integral mult for PID path follower
PID_D = -0.2                        # differential mult for PID path follower
PID_THROTTLE = 0.2                  # constant throttle value during path following
SAVE_PATH_BTN = "cross"             # joystick button to save path
RESET_ORIGIN_BTN = "triangle"       # joystick button to press to move car back to origin

# Intel Realsense D435 and D435i depth sensing camera
REALSENSE_D435_RGB = True       # True to capture RGB image
REALSENSE_D435_DEPTH = True     # True to capture depth as image array
REALSENSE_D435_IMU = False      # True to capture IMU data (D435i only)
REALSENSE_D435_ID = None        # serial number of camera or None if you only have one camera (it will autodetect)

# Stop Sign Detector
STOP_SIGN_DETECTOR = False
STOP_SIGN_MIN_SCORE = 0.2
STOP_SIGN_SHOW_BOUNDING_BOX = True
STOP_SIGN_MAX_REVERSE_COUNT = 10    # How many times should the car reverse when detected a stop sign, set to 0 to disable reversing
STOP_SIGN_REVERSE_THROTTLE = -0.5     # Throttle during reversing when detected a stop sign

# FPS counter
SHOW_FPS = False
FPS_DEBUG_INTERVAL = 10    # the interval in seconds for printing the frequency info into the  @shell

Ezward, 2022-03-22:
config has `DONKEY_GYM = False`, so it this the actual config you are using?  I don't think so because this is all uncommented; I suspect this is config.py.  Are you modifying config.py or myconfig.py? Please update the issue with contents of the actual myconfig.py file that you are using.

Philan014, 2022-03-22:
ok now i made it even worse...

Ezward, 2022-03-22:
Your myconfig.py look fine. When you try to record, where are you looking for the images (exact path).  The tub format has changed, so they won't be in the same place as when you used Donkeycar 3.1

Philan014, 2022-03-22:
home/user/mycar/data/images is empty after trying to record data. If i turn on the option to save the tubs each time after the donkey is started in the config, the same thing happens. no images appear in the folder of the tubs.
I tried uninstalling miniconda3 and then reinstalling it along with recreatind donkey environment. same thing happens. It might me related to the #996 but i still have no idea how to solve it : <

Ezward, 2022-03-23:
It's possible it is a bug that got added with this pull request https://github.com/autorope/donkeycar/pull/985  That PR was to make the webui and the game controllers play nice together.  There was a lot of state passed around that had to be synchonized; it's possible I messed up the recording state.

Philan014, 2022-03-23:
fixed in #996 there are still issues with the record on throttle in ps4 cotroller but now its ok :)

closed 2022-03-23 by Philan014