autorope/donkeycar #1005: 4.3.5.1 is not recording data

Ezward, 2022-04-03:
version: 4.3.6.1
RaspberryPi 4
Relevant config:
- `CAMERA_TYPE = "PICAM"`
- `DRIVE_TRAIN_TYPE = "PWM_STEERING_THROTTLE"`
- `CONTROLLER_TYPE = 'xbox'`
- `JOYSTICK_MAX_THROTTLE = 1 `
- `AUTO_RECORD_ON_THROTTLE = True`
- `JOYSTICK_DEADZONE = 0.1`
- `RECORD_DURING_AI = False`
- `AUTO_CREATE_NEW_TUB = False`

Ran with a fresh install of 4.3.6.1.  I had auto-save with throttle on; the configuration in myconfig.py that causes recording whenever the throttle is non-zero: `AUTO_RECORD_ON_THROTTLE = True`.  I drove a number of laps to create new data.  during that time I used the 'x' button to delete mistakes.  The results was that it created the catalog and images folder, but collected no data.  **The catalog was zero bytes and the images folder was empty.**

zlite, 2022-04-04:
I'm running 4.3.8 and it's saving the data correctly for me

DocGarbanzo, 2022-04-04:
Thanks @zlite - @Ezward, this is your #989 which fixes it then. That's the only 'real' difference between the two version (i.e. HEAD of main and 4.3.6.1)

Ezward, 2022-04-05:
This is actually a duplicate of another issue and that issue has a proposed fix https://github.com/autorope/donkeycar/issues/996#issuecomment-1075800626

Ezward, 2022-04-05:
> I'm running 4.3.8 and it's saving the data correctly for me

@zlite What controller are you using when you test with 4.3.8?

DocGarbanzo, 2022-04-05:
@Ezward - do you suggest that I should use the patch in #996 in the 4.3.6 branch instead of PR #989? I can do that.

Ezward, 2022-04-07:
The problem is addressed by this pull request https://github.com/autorope/donkeycar/pull/1007 which is now merged.

Ezward, 2022-04-10:
I'm reopening this issue.  I've tested with 4.6.10 after a user complained on discord that this was still broken.  They are correct, AUTO_RECORD_ON_THROTTLE is broken; it does not record when that is True and the game controller is used to drive; it creates the metadata files, but they are zero length and it creates the images folder, but is remains empty.  Also toggling manual recording with the game controller when AUTO_RECORD_ON_THROTTLE is on is broken.  I am able to toggle recording using the webui.  I am unable to toggle recording using the game controller.  It does get the button press, but it will not toggle the mode.  someone had tested the fix and said it worked, but apparently that is not true.  In addition, if I set AUTO_RECORD_ON_THROTTLE = False, the the game controller button says it turned on recording, but it can never toggle it off.  Also in that condition, the game controller never records any data, but the webui will if you use the virtual joypad.  So it is more than toggling the mode that is broken.

closed 2022-05-03 by Ezward