autorope/donkeycar #1006: PWM_STEERING_THROTTLE generates I2C errors

zlite, 2022-04-04:
For some reason the new PWM_STEERING_THROTTLE parameter, which replaces `pigpio_pwm` is trying to call the I2C driver, which creates a conflict if that's already being used for the OLED screen. This option should not be using I2C at all AFAIK

[image]

Ezward, 2022-04-04:
Here is what I think is happening.  We are using some older packages for PCA9685 and newer ones for the OLED screen 
```
              'Adafruit_PCA9685',
              'adafruit-circuitpython-lis3dh',
              'adafruit-circuitpython-ssd1306',
```
The older PCA9685 library uses a different dependent library to talk to the I2C bus and I believe it is conflicting with the circuit python library that does the same thing.  There is a newer PCA9685 library that plays nice https://github.com/adafruit/Adafruit_CircuitPython_PCA9685

We probably need to change out our older implementation to the newer one.

Ezward, 2022-04-04:
@zlite can you share your full PWM_STEERING_THROTTLE configuration here?

zlite, 2022-04-06:
I didn't change anything. Just used the defaults (and it failed with that I2C error)

Ezward, 2022-04-06:
> 

So @zlite you are using a standard donkeycar with PCA9685 and you have also enabled the OLED `USE_SSD1306_128_32 = False` ?  I'm tryting to get to a place where I can repeat the issue and I'm not there yet.

zlite, 2022-04-06:
No, I'm not using a PC9685. I'm using the RC Hat with the pigpio driver

Ezward, 2022-04-06:
@zlite  The defaults in config.py are for a standard donkeycar with PCA9685.  You will want to use the configuration described in`software setup` link.  I used that setup with my RC Hat and it works well.

That being said, the OLED should work on a standard donkeycar with PCA9685 and it seems it does not.  I'll try to reproduce that error on my test rig.

zlite, 2022-04-07:
Okay, now I'm totally confused. This is what the myconfig.py file says:
[image]

So I did what it said, and used PWM_STEERING_THROTTLE. And then got a lot of I2C (thus PCA9685) errors. Why is an I2C servo driver board part of a GPIO PWM library???  GPIO PWM is supposed to be *instead* of I2C servo boards

Ezward, 2022-04-07:
Output pins can be sourced from GPIO (using either the RPi.GPIO libarary or the PiGPIO) library and/or they can be sourced via a PCA9685.  We use pin specifiers that define where the signal is sourced.  So a pin specifier is a string that specifies 1) the pin source (pin provider) 2) parameters particular to the pin provider 3) the pin number.  so "PCA9685.0:40.1" specifies a pin 1 on the PCA9685 on bus 0 at address 0x40.  "RPI_GPIO.BOARD.13" specifies the a gpio board pin number 13 using the RPi.GPIO library.  "PIGPIO.BWM.12" specifies gpio pin number 13 using Broadcom numbering scheme sourced using the PiGPIO libary.  We may expose output pins in this way via arduino in the future.  A given drive train can source pins in different ways without requiring separate drivers.  That is explained in the docs, but clearly it is a little too hidden.  It can be found here. https://docs.donkeycar.com/parts/pins/

So what would help is if you could attach your configuration that you are using to this issue, then I can provide very specific advice and we can fixup the docs to make sure no one else gets confused (you are not the first by the way, so I know we need to do some work on this).

zlite, 2022-04-07:
My config is attached, which is basically just the default. 

I understand that you're saying you've merged the I2C and GPIO modules, and now I should be specifying those pins as follows
[image]

I can do that, but this whole process has made using PWM quite a bit more complicated (and, as you say, poorly documented)

Isn't there an easier way for people to just specify whether they want to use I2C or GPIO and let the defaults do the rest of the work?

[myconfig.txt](https://github.com/autorope/donkeycar/files/8447068/myconfig.txt)

zlite, 2022-04-08:
BTW, making the above change results in a different conflict (now with my I2C OLED screen:
[image]

Ezward, 2022-04-09:
The error is saying that there are mixed pin addressing modes in use; Adafruit Blinka is using BCM numbering, the drivetrain is using board number.  You could switch BCM addressing for your drivetrain; that would solve that conflict.  Instead, I would use PiGPIO library for the steering pin and throttle pin output.   You are already using it for input pins for reading the receiver and you would use it if you were reading an encoder, so you should just use it everywhere.  PIGPIO required BCM numbering, so you will need to switch the pin numbers to reflect that addressing scheme.

zlite, 2022-04-10:
Switching to BCM numbering for the PWM output did resolve the OLED problem, but only the steering pin (BCM 13) works and even it has a very different center position from the defaults we had with pigpio (the steering is offset 50% to the left). The throttle pin (18) does not. Both work normally with pigpio on the same pins. 
[image]