autorope/donkeycar #1008: Creating 'donkey' environment from ubuntu.yml takes forever

nabanita-, 2022-04-07:
I am trying to setup the donkey environment in my desktop (Ubuntu18.04.6 LTS) following the instructions [here](https://docs.robocarstore.com/guide/host_pc/setup_ubuntu/). But solving environment is taking forever. 
[image]

I also tried by setting channel_priority as strict (`conda config --set channel_priority strict`) but it resulted in a huge number of conflicts, so I ditched that route.
 

Is there something that I am missing? I am using the yaml file in the main branch of this github repository.

c1505, 2022-04-07:
@nabanita- I don't know why the conda install takes so long, but you aren't alone in that experience of it taking a very long time or hanging.  I think mamba should be the recommended install .  There is a PR for docs changes for that here that I reviewed https://github.com/autorope/donkeydocs/pull/37 .  I don't have write access to merge it though.  

Just noticing that the docs you are using are on a separate domain and use a separate github repository than what I have used https://docs.donkeycar.com/ .  Both the code and docs have not been updated in about 9 months for the robocarstore org.  I am not sure who has access to that repo and how much they are in the donkeycar discord.  I would suggest using this repository for code and the previously mentioned link for docs.

Ezward, 2022-04-07:
I have an experimental branch that has a bash script for installing on unbuntu.  Once you clone/checkout the branch, the run `./install.ubuntu.pip.sh`  https://github.com/autorope/donkeycar/tree/990-ubuntu-install-script

victorjnpires, 2022-04-10:
@nabanita- I've also had issues with conda taking forever on conflict resolution, and mamba was the solution. This is my simplified yml:

```
name: donkey

channels:
  - conda-forge
  - defaults

dependencies:
  - python=3.7
  - h5py
  - pillow
  - opencv
  - matplotlib
  - tornado
  - docopt
  - pandas
  - pylint
  - pytest
  - pytest-cov
  - codecov
  - pip
  - imgaug
  - progress
  - moviepy
  - paho-mqtt
  - PrettyTable
  - pyfiglet
  - mypy
  - numpy
  - psutil
  - kivy=2
  - plotly
  - pyyaml
  - tensorflow-gpu=2.7
  - pip:
    - git+https://github.com/autorope/keras-vis.git
    - simple-pid
    - opencv-python-headless
```

nabanita-, 2022-04-13:
> @nabanita- I don't know why the conda install takes so long, but you aren't alone in that experience of it taking a very long time or hanging. I think mamba should be the recommended install . There is a PR for docs changes for that here that I reviewed [autorope/donkeydocs#37](https://github.com/autorope/donkeydocs/pull/37) . I don't have write access to merge it though.
> 
> Just noticing that the docs you are using are on a separate domain and use a separate github repository than what I have used https://docs.donkeycar.com/ . Both the code and docs have not been updated in about 9 months for the robocarstore org. I am not sure who has access to that repo and how much they are in the donkeycar discord. I would suggest using this repository for code and the previously mentioned link for docs.

I hadn't noticed the difference in domains! Thank you, I will keep that in mind. Solved my problem ^.^

DocGarbanzo, 2022-10-27:
The PR https://github.com/autorope/donkeydocs/pull/37 got merged, so closing this now.

closed 2022-10-27 by DocGarbanzo