autorope/donkeycar #1010: Missing panda for manage.py drive

Benvii, 2022-04-19:
Hi,
I follow the documentation, I noticed there is a missing dependency when doing a raspberry pi installation, I have installed `pi` extra dependencies with `pip install -e .[pi]` (as [mentioned here](https://github.com/autorope/donkeydocs/blob/master/docs/guide/robot_sbc/setup_raspberry_pi.md?plain=1#L227)).

But when I try to launch `python manage.py drive --js` I got a missing dependency `pandas`. Pandas seems to be included in the extras_require of pc [setup.py#L74](https://github.com/autorope/donkeycar/blob/main/setup.py#L74).

- Do we really need it on raspberry pi ?
- Should it be included in `pi` extras_require, dependency set ?

Thanks in advance

```
python manage.py drive --js
________             ______                   _________              
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /    
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/     
                                 /____/                              

using donkey v4.3.10 ...
WARNING:donkeycar.parts.pins:pigpio was not imported.
Traceback (most recent call last):
  File "manage.py", line 27, in <module>
    from donkeycar.parts.datastore import TubHandler
  File "/home/pi/projects/donkeycar/parts/datastore.py", line 19, in <module>
    import pandas as pd
ModuleNotFoundError: No module named 'pandas'
```

DocGarbanzo, 2022-05-23:
Addressed in #1015.

DocGarbanzo, 2022-05-23:
Merged to `main`.

closed 2022-05-23 by DocGarbanzo