autorope/donkeycar #1016: OSError: [Errno 121] Remote I/O error while calibrating

fswalden, 2022-05-27:
When initially calibrating the car, I receive this error "OSError: [Errno 121] Remote I/O error," the meaning of which is unclear to me. I am using a Jetson nano, and the address presented using "sudo i2cdetect -r -y 1" is the same address as the one I put in my myconfig.py folder. Any advice on possible causes or solutions to this problem would be greatly appreciated.

fswalden, 2022-05-27:
Now I'm getting a different error, "RuntimeError: Could not determine default I2C bus for platform." While I might have other issues, it's currently not recognizing that I edited the myconfig.py file. Is there something I could be doing wrong there?

TCIII, 2022-05-28:
@fswalden,

Did you set "PCA9685_I2C_BUSNUM = 1" since you are using a Nano? 
If you did and you are still getting the I/O error, then you may have a defective PCA9685 servo controller or a bad I2C connections. 

TCIII

fswalden, 2022-05-29:
> @fswalden,
> 
> Did you set "PCA9685_I2C_BUSNUM = 1" since you are using a Nano? If you did and you are still getting the I/O error, then you may have a defective PCA9685 servo controller or a bad I2C connections.
> 
> TCIII

Yeah, I set the bus number and the address number. However, when I look through the error messages and warnings, it seems like it isn't recognizing that. Am I missing something in terms of getting my myconfig.py to have its code reflected in the output? Sorry, I'm a bit new to all this. Thanks!

TCIII, 2022-05-29:
@fswalden,

If `sudo i2cdetect -r -y 1` detected the address of your PCA9685, then your I2C connection as well as the health of the PCA9685 should not be suspect.
What I2C pin numbers are you using on your Nano?
What version of DC are you using? Are you using the "master" branch?
Does your myconfig.py file look like this:
PCA9685_I2C_ADDR = 0x40  (or whatever the hex address should be)
PCA9685_I2C_BUSNUM = 1
and not this:
# PCA9685_I2C_ADDR = 0x40  (or whatever the hex address should be)
# PCA9685_I2C_BUSNUM = 1

TCIII

fswalden, 2022-05-29:
@TCIII 

The required code is not commented, no. I used the nano command to edit it and then used the write function to save the changes on myconfig.py. The circuit board does show up with an address of 0x17, which I put in my code. It's probably something really simple I'm doing wrong because I've never worked with this type of thing before, but nonetheless I don't know what that little thing is yet, so I'm pretty desperate. I'm in the master branch, and I'm not sure the answer to the first question, but I'll have a look on Tuesday when I have access to it again. Thanks for the advice, I'll look into it as much as possible. Good to know that the circuit board should be working.

TCIII, 2022-05-29:
@fswalden,

The stock PCA9685 has a default address of 0x40 so I suspect that you do not have a PCA9685 servo controller and that is why you are getting the I/O errors as the DC software is expecting to communicate with a PCA9685.

TCIII

fswalden, 2022-05-29:
@TCIII 

I did not assemble the car, so I'm kind of having to piece it together. You're probably right about that. Do you know of a way around it, given that I know the bus number and the address?

TCIII, 2022-05-29:
@fswalden,

I really can't help without knowing what the servo controller is in your car.
Please give me a link to the car you purchased so I can see what was used.

TCIII

fswalden, 2022-05-31:
I can't really find anything that says exactly what the controller is. The car is the XIAOR Geek XR-F2 Donkeycar with Jetson Nano. It's very difficult to get any information since I didn't buy the car and much of the online information I've found is in Chinese.

TCIII, 2022-05-31:
@fswalden,

They appear to be using a BTS7960 motor controller chip which is really not a typical ESC used in R/C cars.

Here is a [link](https://www.xiaorgeek.com/Study/Study/catalog/cid/32) to their tutorials.

I suspect that if you follow their tutorial for getting this car up and running with the DC software you should be okay.

Regards,
TCIII

fswalden, 2022-05-31:
@TCIII 

Thank you!

TCIII, 2022-05-31:
@fswalden,

I think that you will find in the tutorial that you have to use the XIOR version of donkey car and not the one from the DC website.

Regards,
TCIII

Ezward, 2023-02-18:
I'm going to close this.  The car is not supported by mainline Donkeycar.  Support will have to come from the manufacturer.

closed 2023-02-18 by Ezward